olikraus/u8g2 #2195: Invert Reset pin output, for Schmitt trigger buffered pins

alextrical, 2023-06-03:
I'm looking to use a pin to reset a SSD1309 display, however on the motherboard I'm using (an Big Tree Tech EBB42, pin PD3) it is behind a SN74LVC1G17 Schmitt-Trigger Buffer
Unfortunately this inverts the state of the output signal. 
Is there currently a way to request the signal for reset is sent out inverted from its normal, for situations like this?

Kind regards

alextrical, 2023-06-03:
It looks like adding an exclamation mark before the defined pin resolves the issue for me.
i.e. U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ !PD3);

closed 2023-06-03 by alextrical

olikraus, 2023-06-03:
The "!" is a C operatore, which negates the provided pin number. As a consequence u8g2 will try to modify the unintended GPIO with the Arduino number 0 (which is probably D0)

It is probably better to use U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0) so that u8g2 will not touch the reset line at all.

> Is there currently a way to request the signal for reset is sent out inverted from its normal, for situations like this?

No, however you could ask U8g2 to ignore the reset line (with the constructor above) and do the GPIO changes manually before u8g2.begin().

alextrical, 2023-06-03:
> No, however you could ask U8g2 to ignore the reset line (with the constructor above) and do the GPIO changes manually before u8g2.begin().

Thank you.
That's a really elegant way to do it, I will alter my code to do that instead then, as the plan is to try and support SSD1306,SSD1309,SH1106 and SH1107 displays with as little input from an end user as possible for a Filament extruder