olikraus/u8g2 #2196: mui won't draw Label in utf-8

njuFerret, 2023-06-04:
`muif_list `and `fds_data ` both taken from example `MUIMessage.ino`

`muif_t muif_list[] = {  
  MUIF_U8G2_LABEL(),
  MUIF_VARIABLE("BN", &message_result, mui_u8g2_btn_exit_wm_fi)
};

fds_t fds_data[] = 
MUI_FORM(1)
MUI_LABEL(5,10, "Hello U8g2")
MUI_XYAT("BN",20, 30, " Ok ")
MUI_XYAT("BN",70, 30, " Cancel ")
;`

Everything is fine except for when I changed the label text with UTF-8 encoding, such as  `MUI_LABEL(5,10, "你好")` ,
In this case, the label won't draw, while button text with UTF-8 encoding works perfectly.

It seems  `MUIF_U8G2_LABEL `macro register `mui_u8g2_draw_text` to draw label text , where `u8g2_DrawStr` called. 
`u8g2_DrawStr` function replaced by `u8g2_DrawUTF8` function, text with utf-8 is drawn properly.

olikraus, 2023-06-04:
Good point... I need to look into this

olikraus, 2023-06-06:
I have updated the code (not tested yet). Maybe you can takeover the file manually until next release to solve your issue.

njuFerret, 2023-06-08:
ye, the issue has been fixed, same as this

> I have updated the code (not tested yet). Maybe you can takeover the file manually until next release to solve your issue.
> 
> https://github.com/olikraus/u8g2/blob/1f4c239165690bb0dd1561b26c1b7f740745be50/csrc/mui_u8g2.c#L405

closed 2023-06-12 by olikraus