olikraus/u8g2 #2197: Wide.HK SH1122 Oled on Adafruit KB2040 Kee Boar

JasminMelni1990, 2023-06-07:
Hello,
I have a SH1122 display from Wide.HK and i want to use it with the Adafruit KB2040 Kee Boar i would like to use the pins on the board rather than the Stemma QT plug but i can't get it to work. 
I have successfully used it on a Teensy 4.1 using the following bits of code:
`#define U8G2_16BIT`
`U8G2_SH1122_256X64_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);
//U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);`

But using the same code on the Kee Boar it doesn't work. 

I know the Kee Boar is not really meant to be used with Arduino but i haven't been able to get the display to work with python either so i'm hoping i can get some help here.

Already thanks in advance!

olikraus, 2023-06-10:
Not sure how I can help here...

JasminMelni1990, 2023-06-10:
Judging by that i would assume that the display specific code is correct and that my issue has to do something with the Kee Boar specefic code. At least i can narrow it down.