olikraus/u8g2 #2198: How to use gb2312 in C style

songzhishuo, 2023-06-07:
I tried using gb2312 font under C code, but it doesn't display. “u8g2_DrawStr”、“u8g2_DrawUTF8”、“u8g2_DrawGlyph” can not display。 could you tell me whitch API can display GB2312 font

olikraus, 2023-06-07:
Code? Any Error Message?

njuFerret, 2023-06-08:
> I tried using gb2312 font under C code, but it doesn't display. “u8g2_DrawStr”、“u8g2_DrawUTF8”、“u8g2_DrawGlyph” can not display。 could you tell me whitch API can display GB2312 font

If you simply need to display Chinese characters, it's a good idea to save the source in UTF-8 encoding and use `u8g2_DrawUTF8` to draw the characters.

njuFerret, 2023-06-08:
> > > I tried using gb2312 font under C code, but it doesn't display. “u8g2_DrawStr”、“u8g2_DrawUTF8”、“u8g2_DrawGlyph” can not display。 could you tell me whitch API can display GB2312 font
> > 
> > 
> > If you simply need to display Chinese characters, it's a good idea to save the source in UTF-8 encoding and use `u8g2_DrawUTF8` to draw the characters.
> 
> A magical thing happened. I changed the encoding format of vscode from gb2312 to utf-8, but the font library is in gb2312 format, which can be displayed using API u8g2_DrawUTF8.unbelievable

I'm not sure what "the font library is in gb2312 format" means, but if you are referring to the `u8g2_fonts.c file`, it is possible that the source file does not contain any Chinese characters. Therefore, encoding in utf-8 and gb2312 formats would be the same.

nnd，英语累死了

songzhishuo, 2023-06-08:
> I'm not sure what "the font library is in gb2312 format" means, but if you are referring to the `u8g2_fonts.c file`, it is possible that the source file does not contain any Chinese characters. [...]

啊，你是国人啊，那用中文吧哈哈。我是用的b站甘草大佬的gb2312的库

njuFerret, 2023-06-08:
> 

额，这世界，还有谁关心gb2312？？？

songzhishuo, 2023-06-08:
> 界，还有谁关心gb2312？？

为啥？我觉得2312很好用啊。

njuFerret, 2023-06-08:
> > 界，还有谁关心gb2312？？
> 
> 为啥？我觉得2312很好用啊。

gb2312是乱码的根源，当然，微软也是有锅要背的

songzhishuo, 2023-06-08:
大佬是国内维护u8g2的开发者吗？有没有比较全的u8g2的utf8的库。其次为啥u8g2官方没提供mui 相关的例程？是不好用吗？

songzhishuo, 2023-06-08:
> > 
> 
> 不是什么大佬，mui例程有，arduino版本的，不过还不是很熟悉，正在看文档，写的很好，就是看得很累

大佬方便传递个邮箱什么的然后我加下你微信吗？

songzhishuo, 2023-06-08:
> > 
> 
> 都这个时间了，你居然还在水？
> 
> >

这不是遇到了大佬，所以激动的睡不着

njuFerret, 2023-06-08:
> 

> > > 
> > 
> > 
> > 都这个时间了，你居然还在水？
> > >
> 
> 这不是遇到了大佬，所以激动的睡不着

闪....

ziqian-L, 2023-06-14:
我遇到了相同的问题，相同的办法解决的。
果然，GB2312太难用了。
恨来自瓷器😡

piogrammer, 2023-07-10:
我靠~~
同样的问题
鸡巴CUBEIDE,使用中文的话,必须要把工程设置为GBK,否则乱码乱出天际
但是这样的话,u8g2_DrawUTF8就不能显示中文
当然用keil mdk可以解决这个问题,问题是MDK调试freertos又不如cubdide友好
看来只能等作者更新这个bug了