olikraus/u8g2 #2199: Porting U8G2 to CH32V303 with JLX12864 (ST7567) can not init

inows, 2023-06-09:
Chip: CH32V303RCT6
LCD: JLX12864 st7567 
Communication: Hardware SPI

actuality:  lcd can work with original code, but can not work by u8g2, they use the same SPI Init Code, please help me~ thx

SPI Init Code
`
void u8g2_hw_spi_init(void)
{

    GPIO_InitTypeDef    GPIO_InitStructure = {0};
    SPI_InitTypeDef     SPI_InitStructure = {0};

    RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOA, ENABLE);

    if(SPI1 == U8G2_SPI)
    {
        printf("Open SPI1 Clock...\r\n");
        RCC_APB2PeriphClockCmd(U8G2_SPI_CLOCK, ENABLE);
    }else {
        printf("Open SPI2/3 Clock...\r\n");
        RCC_APB1PeriphClockCmd(U8G2_SPI_CLOCK, ENABLE);
    }

    GPIO_InitStructure.GPIO_Pin = PIN_LCD_CS | PIN_LCD_RS | PIN_LCD_RST;
    GPIO_InitStructure.GPIO_Mode = GPIO_Mode_Out_PP;
    GPIO_InitStructure.GPIO_Speed = GPIO_Speed_50MHz;
    GPIO_Init(GPIOA, &GPIO_InitStructure);

    GPIO_InitStructure.GPIO_Pin = PIN_LCD_SCK | PIN_LCD_MOSI;
    GPIO_InitStructure.GPIO_Mode = GPIO_Mode_AF_PP;
    GPIO_InitStructure.GPIO_Speed = GPIO_Speed_50MHz;
    GPIO_Init(GPIOA, &GPIO_InitStructure);

    GPIO_InitStructure.GPIO_Pin = PIN_LCD_MISO;
    GPIO_InitStructure.GPIO_Mode = GPIO_Mode_IN_FLOATING;
    GPIO_Init(GPIOA, &GPIO_InitStructure);

    SPI_InitStructure.SPI_Direction = SPI_Direction_1Line_Tx;
    SPI_InitStructure.SPI_Mode = SPI_Mode_Master;
    SPI_InitStructure.SPI_DataSize = SPI_DataSize_8b;
    SPI_InitStructure.SPI_CPOL = SPI_CPOL_High;
    SPI_InitStructure.SPI_CPHA = SPI_CPHA_2Edge;
    SPI_InitStructure.SPI_NSS = SPI_NSS_Soft;
    SPI_InitStructure.SPI_BaudRatePrescaler = SPI_BaudRatePrescaler_8;
    SPI_InitStructure.SPI_FirstBit = SPI_FirstBit_MSB;
    SPI_InitStructure.SPI_CRCPolynomial = 7;
    SPI_Init(U8G2_SPI, &SPI_InitStructure);

    SPI_Cmd(U8G2_SPI, ENABLE);

    u8 i = 0;

    while(SPI_I2S_GetFlagStatus(U8G2_SPI, SPI_I2S_FLAG_TXE) == RESET)
    {
        i++;
        if(i > 200)
            break;
    }

    printf("U8G2 SPI Initialized!\r\n");
}
`

Original Code
`
#include "lcd.h"
#include "stdlib.h"
#include "lcdfont.h"  	 

//=============transfer command to LCM===============
void transfer_command_lcd(int data)
{
#ifndef USE_HARDWARE_SPI
	u8 i;
	LCD_CS=0;
	LCD_RS=0;
	for(i=0;i<8;i++)
	{
		LCD_SCL=0;
		if(data1&0x80) LCD_SDA =1;
		else LCD_SDA =0;
		LCD_SCL=1;
		data1=data1<<=1;
	}
#else
	GPIO_WriteBit(PORT_LCD_CS, PIN_LCD_CS, 0);
	GPIO_WriteBit(PORT_LCD_RS, PIN_LCD_RS, 0);
#ifdef ENABLE_U8G2_DEBUG
    printf("Send Command Byte: %02X\t", data);
#endif
	SPI_ReadWriteByte(U8G2_SPI, data);
	GPIO_WriteBit(PORT_LCD_CS, PIN_LCD_CS, 1);
#endif
}

//-----------transfer data to LCM---------------
void transfer_data_lcd(int data)
{
#ifndef USE_HARDWARE_SPI
	u8 i;
	LCD_CS=0;
	LCD_RS=1;
	for(i=0;i<8;i++)
	{
		LCD_SCL=0;
		if(data1&0x80) LCD_SDA=1;
		else LCD_SDA=0;
		LCD_SCL=1;
		data1=data1<<=1;
	}
#else
	GPIO_WriteBit(PORT_LCD_CS, PIN_LCD_CS, 0);
	GPIO_WriteBit(PORT_LCD_RS, PIN_LCD_RS, 1);
#ifdef ENABLE_U8G2_DEBUG
	printf("Send Date Byte: %02X\t", data);
#endif
	SPI_ReadWriteByte(U8G2_SPI, data);
	GPIO_WriteBit(PORT_LCD_CS, PIN_LCD_CS, 1);
#endif
} 

void initial_lcd(void)
{
#ifndef USE_HARDWARE_SPI
  GPIO_InitTypeDef  GPIO_InitStructure;
 	
 	RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOA|RCC_APB2Periph_GPIOB|RCC_APB2Periph_GPIOC, ENABLE);	 //使能PA,B,C端口时钟 （ENABLE开启）

	GPIO_InitStructure.GPIO_Pin = GPIO_Pin_4|GPIO_Pin_5;//PC4，PC5推挽输出  
 	GPIO_InitStructure.GPIO_Mode = GPIO_Mode_Out_PP;  //推挽输出
	GPIO_InitStructure.GPIO_Speed = GPIO_Speed_50MHz; //速度50MHz
 	GPIO_Init(GPIOC, &GPIO_InitStructure);	  		    //初始化GPIOC3
 	GPIO_SetBits(GPIOC,GPIO_Pin_4|GPIO_Pin_5);        //PC4，PC5输出高
	
 	GPIO_InitStructure.GPIO_Pin =GPIO_Pin_6|GPIO_Pin_7;   //PA6，PA7    
	GPIO_InitStructure.GPIO_Mode = GPIO_Mode_Out_PP;  //推挽输出
 	GPIO_Init(GPIOA, &GPIO_InitStructure);
 	GPIO_SetBits(GPIOA,GPIO_Pin_6|GPIO_Pin_7);                 

 	GPIO_InitStructure.GPIO_Pin = GPIO_Pin_0;				 //PB0
	GPIO_InitStructure.GPIO_Mode = GPIO_Mode_Out_PP; 
 	GPIO_Init(GPIOB, &GPIO_InitStructure);
 	GPIO_SetBits(GPIOB,GPIO_Pin_0);						      

#endif

 	GPIO_WriteBit(PORT_LCD_RST, PIN_LCD_RST, 0);
	delay_ms(50);
	GPIO_WriteBit(PORT_LCD_RST, PIN_LCD_RST, 1);
	delay_ms(100);

	transfer_command_lcd(0xe2);	 /*软复位*/
	delay_ms(5);
	transfer_command_lcd(0x2c);  /*升压步聚1*/
	delay_ms(5);	
	transfer_command_lcd(0x2e);  /*升压步聚2*/
	delay_ms(5);
	transfer_command_lcd(0x2f);  /*升压步聚3*/
	delay_ms(5);
	transfer_command_lcd(0x24);  /*粗调对比度，可设置范围0x20～0x27*/
	transfer_command_lcd(0x81);  /*微调对比度*/
	transfer_command_lcd(0x1b);  /*0x1a,微调对比度的值，可设置范围0x00～0x3f*/
	transfer_command_lcd(0xa2);  /*1/9偏压比（bias）*/
	transfer_command_lcd(0xc8);  /*行扫描顺序：从上到下*/
	transfer_command_lcd(0xa0);  /*列扫描顺序：从左到右*/
	transfer_command_lcd(0x60);  /*起始行：第一行开始*/
	transfer_command_lcd(0xaf);  /*开显示*/
	
}

void lcd_address(u8 page,u8 column)
{
	page = page - 1;
	transfer_command_lcd(0xb0 + page);   			        //设置页地址。每页是8行。一个画面的64行被分成8个页。我们平常所说的第1页，在LCD驱动IC里是第0页，所以在这里减去1*/
	transfer_command_lcd(((column >> 4) & 0x0f) + 0x10);	//设置列地址的高4位
	transfer_command_lcd(column & 0x0f);				    //设置列地址的低4位
}

/*全屏清屏*/
void clear_screen(void)
{
	unsigned char i,j;
 	for(i=0;i<9;i++)
	{
		lcd_address(1+i,1);
		for(j=0;j<132;j++)
		{
			transfer_data_lcd(0x00);
		}
	}
}

/*显示128x64点阵图像*/
void display_graphic_128x64(u8 page,u8 column,u8 *dp)
{
	int i,j;
	for(j=0;j<8;j++)
	{
		lcd_address(page+j,column);
		for(i=0;i<128;i++)
		{
			transfer_data_lcd(*dp);   
			dp++;
		}
	}          
}

/*显示32x32点阵图像、汉字、生僻字或32x32点阵的其他图标*/
void display_graphic_32x32(u8 page,u8 column,u8 *dp)
{
	u8 i,j;
	for(j=0;j<4;j++)
	{
		lcd_address(page+j,column);
		for (i=0;i<31;i++)
		{	
			transfer_data_lcd(*dp);		/*写数据到LCD,每写完一个8位的数据后列地址自动加1*/
			dp++;
		}
	}
}

/*显示16x16点阵图像、汉字、生僻字或16x16点阵的其他图标*/
void display_graphic_16x16(u8 page,u8 column,u8 *dp)
{
	u8 i,j;
	for(j=0;j<2;j++)
	{
		lcd_address(page+j,column);
		for (i=0;i<16;i++)
		{	
			transfer_data_lcd(*dp);		/*写数据到LCD,每写完一个8位的数据后列地址自动加1*/
			dp++;
		}
	}
}
`

U8G2 Init Code
`
u8g2_Setup_st7567_jlx12864_f(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, u8x8_ch32_gpio_and_delay);

u8 SPI_ReadWriteByte(SPI_TypeDef *SPIx, u8 TxData)
{
#ifdef ENABLE_U8G2_DEBUG
    printf("Act Byte: %02X\r\n", TxData);
#endif
    u8 i = 0;

    while(SPI_I2S_GetFlagStatus(SPIx, SPI_I2S_FLAG_TXE) == RESET)
    {
        i++;
        if(i > 200)
            return 0;
    }

    SPI_I2S_SendData(SPI1, TxData);
    i = 0;

    while(SPI_I2S_GetFlagStatus(SPIx, SPI_I2S_FLAG_RXNE) == RESET)
    {
        i++;
        if(i > 200)
            return 0;
    }

    return SPI_I2S_ReceiveData(SPIx);
}

uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,void *arg_ptr)
{
    switch (msg)
    {
        case U8X8_MSG_BYTE_SEND: /*通过SPI发送arg_int个字节数据*/
            /*配置了DMA取消上一行注释即可*/
            for(int i = 0; i < arg_int; i++)
            {
                uint8_t data = *((uint8_t *)(arg_ptr + i));
#ifdef ENABLE_U8G2_DEBUG
                printf("Send Byte: %02X\tSeq: %d/%d\t", data, i + 1, arg_int);
#endif
                SPI_ReadWriteByte(U8G2_SPI, data);
            }
            /*这是CubeMX生成的初始化*/
            break;
        case U8X8_MSG_BYTE_INIT: /*初始化函数*/
            break;
        case U8X8_MSG_BYTE_SET_DC: /*设置DC引脚,表明发送的是数据还是命令*/
            GPIO_WriteBit(PORT_LCD_RS, PIN_LCD_RS,arg_int);
            break;
        case U8X8_MSG_BYTE_START_TRANSFER:
            u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
            u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
            break;
        case U8X8_MSG_BYTE_END_TRANSFER:
            u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
            u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
            u8x8_gpio_SetDC(u8x8, u8x8->display_info->chip_disable_level);
            break;

        default:
            return 0;
    }
    return 1;
}

uint8_t u8x8_ch32_gpio_and_delay(U8X8_UNUSED u8x8_t *u8x8,
    U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int,
    U8X8_UNUSED void *arg_ptr)
{
    switch (msg)
    {
        case U8X8_MSG_GPIO_AND_DELAY_INIT: /*delay和GPIO的初始化，在main中已经初始化完成了*/
//            u8g2_hw_spi_init();
            break;
        case U8X8_MSG_DELAY_MILLI: /*延时函数*/
            Delay_Ms(arg_int);     //调用谁stm32系统延时函数
            break;
        case U8X8_MSG_GPIO_CS: /*片选信号*/ //由于只有一个SPI设备，所以片选信号在初始化时已经设置为常有效
            GPIO_WriteBit(PORT_LCD_CS, PIN_LCD_CS, arg_int);
            break;
        case U8X8_MSG_GPIO_DC: /*设置DC引脚,表明发送的是数据还是命令*/
            GPIO_WriteBit(PORT_LCD_RS, PIN_LCD_RS, arg_int);
            break;
        case U8X8_MSG_GPIO_RESET:
            break;
    }
    return 1;
}
`
U8G2 Init_Display Packet by Console
[image]

U8G2 Init_Display Packet Capture by LA
[image]

inows, 2023-06-09:
Data Packet is valid,  but the init packet was broken.

[image]
[image][image]

inows, 2023-06-09:
i make it work

jlx12864_st7567  init_seq is wrong in u8g2

paste my working code
[image]

[image]

and the init_display packet broken that i miss this line
[image]

olikraus, 2023-06-09:
Hmmm not so clear to me. Do I need to fix something in u8g2? If so, what exactly needs to be fixed?

inows, 2023-06-10:
need u update the st7567_jlx12864_init_seq like me and update U8X8_MSG_DISPLAY_DRAW_TILE like me   only for st7567_jlx12864

olikraus, 2023-06-10:
The change to the DRAW_TILE function does not make sense. Instead it seems that changing the x offset to 1 should solve your problem then: 
https://github.com/olikraus/u8g2/blob/a143183f0d117d57572a3646567a78900b0de75d/csrc/u8x8_d_st7567.c#L256

Did you try the u8g2.setContrast() command with the original constructor?