olikraus/u8g2 #2200: Bar Graph with customized Width

njuFerret, 2023-06-10:
When using a bar graph as an indicator, such as for PWM's duty ratio, it can be challenging to draw the graph at a specific position and width. Modifications has been made in the source code to implement such a scenario, and a git patch file  was generated as well. 
If possible, a pull request can be made to incorporate these changes into the original code.

besides,  when out of the min/max boundary, value will be constrained at the min/max value without cycle.( the duty ratio of PWM should never be 100%->0%)

[Bar_with_fixed_width.patch](https://github.com/olikraus/u8g2/files/11712455/Bar_with_fixed_width.patch)

olikraus, 2023-06-10:
somehow the patch file is lost here. A PR would be cool, I could then look a little bit deeper into your modification.

olikraus, 2023-06-10:
Ah you did this already: https://github.com/olikraus/u8g2/pull/2201

olikraus, 2023-06-11:
reopened as a reminder for myself

olikraus, 2023-06-11:
I have recreated a new field function instead of merging the PR: https://github.com/olikraus/u8g2/wiki/muiref#fixed-width-bar-graph-uint8

closed 2023-06-11 by olikraus

olikraus, 2023-06-12:
* Added NO_WRAP option to both bar graph fields
 * Bar graph with fixed pixel width
 * Redesigned software structure
 * Test case: https://github.com/olikraus/u8g2/blob/master/sys/sdl/mui_issue_2200/main.c