olikraus/u8g2 #2204: SW I2C not working on SH1106

BestBoyBerlin, 2023-06-17:
Hi, I have two sh1106 displays with the same i2c address. so I wanted to use one with hardware i2c and one with software i2c on my Arduino Uno. The hardware one works but i cant get the software i2c to work. I tried with different pins and u8x8 as well as u8g2 but nothing worked. I then tried this library https://github.com/notisrac/SH1106Lib and it worked. so my question is if im using the u8x8 library wrong or if there is a bug in the library. Here is a minimal source code where only the hardware i2c works:

```
#include <U8x8lib.h>

U8X8_SH1106_128X64_NONAME_SW_I2C u8x8(1, 2, U8X8_PIN_NONE);
U8X8_SH1106_128X64_NONAME_HW_I2C u8x8_2(U8X8_PIN_NONE);

void setup() {
    u8x8.begin();
    u8x8_2.begin();
}

void loop(void) {
    u8x8.setFont(u8x8_font_chroma48medium8_r);
    u8x8.drawString(0,0,"Test1!");
    u8x8_2.setFont(u8x8_font_chroma48medium8_r);
    u8x8_2.drawString(0,0,"Test2!");
    delay(1000);
}
```

olikraus, 2023-06-17:
Looks correct. maybe the pin numbers are incorrect.

BestBoyBerlin, 2023-06-17:
This is what I thought at first, so I checked and tried a lot. I changed the order, tried different pin names, used both analog and digital pins, but nothing worked.  Also the pin numbers should be correct, because when I use SH1106lib with the same pins it works. There I define the ports like this:

```
#define SDA_PORT PORTC
#define SDA_PIN 2 
#define SCL_PORT PORTC
#define SCL_PIN 1
```

because this library uses SoftI2CMaster under the hood. So I think there must be some kind of bug in your library or the underlying software I2C library you are using.

olikraus, 2023-06-17:
U8g2/u8x8 require the Arduino pin number

BestBoyBerlin, 2023-06-18:
Thank you for your help. You were right about the pin numbers. I got it to work on the Uno now. But now I have another problem. I tried using the same code on an Arduino Nano 33 BLE and there the SW I2C does not work. So my question is if your library only supports AVR architectures or if I need to do something different on this board. I tried the A1 and A2 pins and tried many different numbers in the code but nothing worked. The display does nothing.

olikraus, 2023-06-18:
U8g2 SW I2C just depends on digitalWrite() from the Arduino Lib, so it should be portable across many platforms as long as digitalWrite() is supported.

Reference: https://reference.arduino.cc/reference/en/language/functions/digital-io/digitalwrite/

BestBoyBerlin, 2023-06-18:
Ok, then things just get weirder. I tested both pins with the digitalWrite() function and a LED, found out the correct pin numbers and saw that digitalWrite() works. But there is still nothing on the display. With HW I2C the display works, so the display is not broken. Do you have any other idea what the problem might be?

olikraus, 2023-06-18:
Maybe a wiring issue, like swapped I2C pins. Sometimes also a reset connection is required.

BestBoyBerlin, 2023-06-18:
I have checked the wiring several times, even swapped the I2C pins, so I dont think the problem is there. Im not sure what you mean by reset pin, but the display only has four pins and the exact same configuration without reset works with both HW I2C and SW I2C on the Arduino Uno. Out of curiosity, I connected an LED to the SW I2C pins and both output a signal. With the clock pin I saw a consistent frequency, but it seemed different on the Uno than on the Nano 33 BLE. Is it possible that the timing on the Nano 33 BLE is wrong because the processor is faster?

olikraus, 2023-06-18:
... I don't own a Nano 33 BLE...

BestBoyBerlin, 2023-06-18:
Ok, so I guess I have to wait until you can reproduce the bug yourself one day, or I somehow find a solution on my own. But thanks anyway for your help in getting it to work on the Uno. I really appreciate your work on the library.

olikraus, 2023-06-18:
Did you try to reduce speed with setBusClock() command?

BestBoyBerlin, 2023-06-19:
I have now tried many different speeds, but unfortunately nothing has changed. The display still shows no reaction.