olikraus/u8g2 #2205: font: "u8g2_font_micropixel_te" missing

HelmiMUC, 2023-06-19:
Hello,

after installing the last version 2.34.22 the font "u8g2_font_micropixel_te" is no more available.

olikraus, 2023-06-19:
hmm strange, thanks for the int.

The font itself is still here: 

https://github.com/olikraus/u8g2/blob/master/tools/font/build/single_font_files/u8g2_font_micropixel_te.c

So you can  just copy and past the c code (https://raw.githubusercontent.com/olikraus/u8g2/master/tools/font/build/single_font_files/u8g2_font_micropixel_te.c)  into your ino file.

Nevertheless it looks like the original file is somehow lost. I need to look for it...

olikraus, 2023-06-19:
I think it was not fully added to the last release. It should have been here: https://github.com/olikraus/u8g2/wiki/fntgrpbitfontmaker2
But I probably decided not to add this font to the last release, but I think it still got included into the last release without being documented.

Font: http://www.pentacom.jp/pentacom/bitfontmaker2/gallery/?id=9254
License: MicroPixel by SpicyGame, license : (Public Domain)

olikraus, 2023-06-19:
@HelmiMUC : Will it be ok to add the font manually for now? It will be part of the next release.

HelmiMUC, 2023-06-19:
i found a better solution: i can use another font

olikraus, 2024-08-24:
the font is part of the build script, so it should be there, closing this issue...

closed 2024-08-24 by olikraus