olikraus/u8g2 #2207: Wire1 errors with 328P and minicore

m-wb, 2023-06-20:
Hi,
I am trying to compile for the Atmega328P using PlatformIO and miniCore.

as soon as I include the U8g2 lib, I get compilation errors regarding Wire (even with empty / minimal examples)
```
#include <U8g2lib.h>
```

I get these errors:
```
Compiling .pio\build\
Upload_ISP\lib253\U8g2\U8x8lib.cpp.o
\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c: In function 'twi_init1':
\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:88:16: error: 'SDA1' undeclared (first use in this function); did you mean 'SDA'?
   digitalWrite(SDA1, 1);
                ^~~~
```

and many more that look the same, ending with 
```
*** [.pio\build\Upload_ISP\libc9a\Wire1\Wire1.cpp.o] Error 1
*** [.pio\build\Upload_ISP\libc9a\Wire1\utility\twi1.c.o] Error 1
```
Google searches led to nothing and I have no idea on how to proceed. I was reminded of the issue I had last year with the 328PB which was solved here: https://github.com/olikraus/u8g2/issues/1890

Interestingly it worked until I installed the library fresh (I had U8g2 2.33.9 installed globally via pio, removed it, then added it again via platformio.ini in lib_deps = olikraus/U8g2@^2.34.22). I'm not 100% positive whether I changed anything in the old installation, but if, then it certainly only was the fix of issue 1890 before it was integrated into the code vie PR.

Any ideas?

olikraus, 2023-06-21:
To me it looks like an issue with the avr-minicore.
Did you try the suggestion from the above refere issue? 
Try to comment / remove this line in your local u8g2 installation:
https://github.com/olikraus/u8g2/blob/74e379c08f4e3db2c9ac18db72cdee0d47276a58/cppsrc/U8x8lib.h#L138

Maybe this will fix the error...

m-wb, 2023-06-21:
Thanks for the answer. Unfortunately, commenting did not help, the error is the same. I'll post the full error response below, in case this might help

`
HARDWARE: ATMEGA328P 8MHz, 2KB RAM, 32KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr-minicore @ 2.2.2 
 - tool-avrdude @ 1.70100.0 (7.1.0) 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 24 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.34.22
Building in release mode
Compiling .pio\build\Upload_ISP\src\main.cpp.o
Compiling .pio\build\Upload_ISP\libc9a\Wire1\Wire1.cpp.o
Compiling .pio\build\Upload_ISP\libc9a\Wire1\utility\twi1.c.o
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c: In function 'twi_init1':
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:88:16: error: 'SDA1' undeclared (first use in this function); did you mean 'SDA'?
   digitalWrite(SDA1, 1);
                ^~~~
                SDA
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:88:16: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:89:16: error: 'SCL1' undeclared (first use in this function); did you mean 'SCL'?
   digitalWrite(SCL1, 1);
                ^~~~
                SCL
In file included from c:\users\marku\.platformio\packages\toolchain-atmelavr\avr\include\avr\io.h:99:0,
                 from C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:25:
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:92:7: error: 'TWSR1' undeclared (first use in this function); did you mean 'TWSR'?
   cbi(TWSR1, TWPS0);
       ^
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:92:3: note: in expansion of macro 'cbi'
   cbi(TWSR1, TWPS0);
   ^~~
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:94:9: error: lvalue required as left operand of assignment
   TWBR1 = ((F_CPU / TWI_FREQ) - 16) / 2;
         ^
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:102:3: error: 'TWCR1' undeclared (first use in this function); did you mean 'TWSR1'?
   TWCR1 = _BV(TWEN) | _BV(TWIE) | _BV(TWEA);
   ^~~~~
   TWSR1
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c: In function 'twi_disable1':
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:114:3: error: 'TWCR1' undeclared (first use in this function); did you mean 'TWCR'?
   TWCR1 &= ~(_BV(TWEN) | _BV(TWIE) | _BV(TWEA));
   ^~~~~
   TWCR
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:117:16: error: 'SDA1' undeclared (first use in this function); did you mean 'SDA'?
   digitalWrite(SDA1, 0);
                ^~~~
                SDA
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:118:16: error: 'SCL1' undeclared (first use in this function); did you mean 'SCL'?
   digitalWrite(SCL1, 0);
                ^~~~
                SCL
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c: In function 'twi_setAddress1':
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:130:3: error: 'TWAR1' undeclared (first use in this function); did you mean 'TWA1'?
   TWAR1 = address << 1;
   ^~~~~
   TWA1
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c: In function 'twi_setFrequency1':
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:141:9: error: lvalue required as left operand of assignment
   TWBR1 = ((F_CPU / frequency) - 16) / 2;
         ^
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c: In function 'twi_readFrom1':
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire1\src\utility\twi1.c:218:9: error: 'TWDR1' undeclared (first use in this function); did you mean 'TWD1'?
         TWDR1 = twi_slarw;
         ^~~~~
[... 96 lines cut ...]
     ^~~~~~~
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire\src/Wire.h:55:5: note:   candidate expects 0 arguments, 15 provided
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire\src/Wire.h:37:7: note: candidate: constexpr TwoWire::TwoWire(const TwoWire&)
 class TwoWire : public Stream
       ^~~~~~~
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire\src/Wire.h:37:7: note:   candidate expects 1 argument, 15 provided
C:\Users\marku\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire\src/Wire.h:37:7: note: candidate: constexpr TwoWire::TwoWire(TwoWire&&)
[... cut ...]
Compiling .pio\build\Upload_ISP\lib253\U8g2\clib\u8x8_d_ssd1320.c.o
Compiling .pio\build\Upload_ISP\lib253\U8g2\clib\u8x8_d_ssd1322.c.o
Compiling .pio\build\Upload_ISP\lib253\U8g2\clib\u8x8_d_ssd1325.c.o
*** [.pio\build\Upload_ISP\libc9a\Wire1\utility\twi1.c.o] Error 1
Compiling .pio\build\Upload_ISP\lib253\U8g2\clib\u8x8_d_ssd1326.c.o
*** [.pio\build\Upload_ISP\libc9a\Wire1\Wire1.cpp.o] Error 1
========================== [FAILED] Took 3.10 seconds ==========================
[Finished in 3.7s]`

olikraus, 2023-06-23:
From my perspective the error is in the board library. So u8g2 is the wrong place to ask.
The only thing what you could do here is to fully disable I2C support:

Remove the following line to disable I2C:
https://github.com/olikraus/u8g2/blob/74e379c08f4e3db2c9ac18db72cdee0d47276a58/cppsrc/U8x8lib.h#L84

m-wb, 2023-06-23:
Thanks for your response. Unfortunately I'll need I2C to drive an OLED. But I'll find another way around this.

Thanks again.

m-wb, 2023-06-25:
FYI: I did compile successfully without any problems on a M0+, so yes, probably an issue with the minicore board lib.

Thanks for the support anyhow :)

closed 2023-06-25 by m-wb

mumrah, 2023-10-13:
@m-wb did you ever get this working with the 328p? I'm running into the same issue https://community.platformio.org/t/cannot-compile-basic-u8g2-sketch/36168

m-wb, 2023-10-13:
Hi,
No, I did not on the 328p. i had to switch to a more capable MCU anyways due to other requirements, so it was no issue for me anymore.

But good to know for future projects, thanks for letting me know.