olikraus/u8g2 #2208: Incompileable on ArduinoDroid

warriorCavt, 2023-06-21:
ArduinoDroid spits out "undefined reference for 'u8g2_[...]' and I can't compile with this library so I'm forced to use Adafruit SH110X, even if I like U8g2 much more. 

U8g2 uses less varmem, and has dozens of fonts that even Adafruit wouldn't imagine. My favorite is Nokia Cellphone, I use it in a lot of projects. And 5x7_mf (basically 4x7) one is good for small letters.

olikraus, 2023-06-21:
U8g2 is tested and works with the Arduino IDE.
If it (still) doesn't work with ArduinoDroid, then ArduinoDroid is not compatible with Arduino.
So ArduinoDroid needs to be fixed.

warriorCavt, 2023-06-25:
I fixed it; moved *.c files to the parent folder and copied *.h also to the parent folder; but it's now recompiling everything everytime
I will move to U8x8 so I use less memory, this fills my needs.
And by the way — how do I delete the last char from a U8x8 log (let's name it 'text')? Can I do it by just editing the text_Buffer? Or what do I have to do? I have added a 'prompt' string to handle the keyboard prompt for commands, I just need a possibility to "backspace"