olikraus/u8g2 #2210: Support for Sharp Memory Display LS011B7DH03

Phasikk, 2023-06-22:
Hello,

I'm currently using a 144x168 pixel LS013B7DH05 Memory Display from Sharp with u8g2 and wanted to physically shrink down a project of mine. The new display I got is a 160x68 pixel LS011B7DH03, which u8g2 currently doesn't support.

I was able to successfully test the LS011B7DH03 display with the Adafruit Memory Display library after setting the proper display size in the constructor. Can I perform a similar change in u8g2 to get the display to work?
Maybe it's also possible to add a fitting constructor to u8g2 based on the ones of existing memory displays, since the Adafruit library only requires me to update the screen size?

I appreciate any help with this topic. Thanks!

olikraus, 2023-06-22:
Is there any datasheet for this display?

Phasikk, 2023-06-23:
Yes, there is: [LS011B7DH03 Datasheet](https://www.sharpsde.com/fileadmin/products/Displays/Specs/LS011B7DH03_24Nov17_Spec_LD-29Y03.pdf).
For reference, the display I got is a nice!view display ([docs](https://nicekeyboards.com/docs/nice-view/)), which I figured was the most accessible way to get a smaller memory display.

Phasikk, 2023-07-23:
Alright, I have probably put a few hours too many into getting the display to work with u8g2. At this point I can get something to show up on the display, although it's far from perfect. Keep in mind that I have very limited knowledge with programming in this kind of depth and complexity and am trying to guess as to what might work.

I managed to find the file that defines the memory display properties, U8g2\src\clib\u8x8_d_ls013b7dh03.c, and decided to change the values in u8x8_ls013b7dh03_128x128_display_info to match my display. I compared the timing values in the data sheet for my 160x68 and the already supported (and working) 168x144 memory display and found no difference, so I copied its values to start with. The only four parameters I that probably need changing are `tile_width`, `tile_height`, `pixel_width `and `pixel_height`.

Testing was done with a Teensy 4.0 and the nice!view LS011B7DH03 display breakout, connected as follows:

| Teensy 4.0 | nice!view LS011B7DH03 |
| --- | --- |
| GND | GND |
| 3.3V | VCC |

Programming was don with VS Code and PlatformIO. After every change the u8x8_d_ls013b7dh03.c file was recompiled according to the log and some changes were noticeable, so I guess that worked.  Testing was done with the full buffer "Hello World" example and this constructor:
`U8G2_LS013B7DH03_128X128_F_4W_HW_SPI u8g2(U8G2_R0, 10, U8X8_PIN_NONE, U8X8_PIN_NONE);`

I found that the best result can be achieved by setting the `tile_width `to 9. Higher or lower values will only show some line fragments but nothing even remotely legible, values over 17 cause the screen to stay blank. The values for the `pixel_width `and `pixel_height `(160 and 68) can seemingly be interchanged without affecting the displayed result in a noticeable way.

. Maybe some pixels are different, but the overall image stays the same. At this point I've reached the end of my limited knowledge, everything to that striped image was an educated guess on my side that somehow resulted in something.

Phasikk, 2023-07-23:
Alright, I have put some more time into getting the display to work with u8g2. At this point I can get something to show up on the display, although it's far from perfect.

I managed to find the file that defines the memory display properties, `U8g2\src\clib\u8x8_d_ls013b7dh03.c`, and decided to change the values in `u8x8_ls013b7dh03_128x128_display_info` to match my display. I compared the timing values in the data sheet for my 160x68 LS011B7DH03 and the already supported (and working) 168x144 LS013B7DH05 memory display and found no difference, so I copied its values to start with. The only four parameters that probably need changing are `tile_width`, `tile_height`, `pixel_width` and `pixel_height`.

Testing was done with a Teensy 4.0 and the [nice!view LS011B7DH03](https://nicekeyboards.com/nice-view/) display breakout with the follwing wiring.

| Teensy 4.0 | nice!view LS011B7DH03 |
| --- | --- |
| GND | GND |
| 3.3V | VCC |
| CS (Pin 10) | CS |
| MOSI (Pin 11)  | MOSI |
| SCK (Pin 13) | SCK |

Everything was tested beforehand with the [Adafruit Sharp Memory Display library](https://github.com/adafruit/Adafruit_SHARP_Memory_Display) and the following code to make sure the setup can work.

```
#include <Adafruit_GFX.h>
#include <Adafruit_SharpMem.h>

#define SHARP_SS   10
#define SHARP_SCK  13
#define SHARP_MOSI 11

Adafruit_SharpMem display(SHARP_SCK, SHARP_MOSI, SHARP_SS, 160, 68);

#define BLACK 0
#define WHITE 1

void setup(void){
  display.begin();
  display.clearDisplay();
  display.refresh();
}

void loop(void){
  display.setTextSize(1);
  display.setTextColor(BLACK);
  display.setCursor(0,0);
  display.write("Hello World!");
  display.refresh();
  delay(500);
}
```
[image]

Testing with u8g2 was done with VS Code and PlatformIO. After every change the `u8x8_d_ls013b7dh03.c` file was successfully recompiled according to the log and some changes were noticeable, so I guess that worked. The code for testing is based on the full buffer "Hello World" example and the constructor for the display data I modified:
`U8G2_LS013B7DH03_128X128_F_4W_HW_SPI u8g2(U8G2_R0, 10, U8X8_PIN_NONE, U8X8_PIN_NONE);`

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

U8G2_LS013B7DH03_128X128_F_4W_HW_SPI u8g2(U8G2_R0, 10, U8X8_PIN_NONE, U8X8_PIN_NONE);

void setup(void){
  u8g2.begin();
}

void loop(void){
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB08_tr);
  u8g2.drawStr(0,10,"Hello World!");
  u8g2.sendBuffer();
  delay(1000);
}
```
I assumed that the `tile_width` and `tile_height` is the respective dimensional resolution divided by 8, based on the display info I found for the other display drivers u8g2 supports. All the displays I looked into have a resolution that is divisible by 8, while the 68px of my display are not. A `tile_width` value of 8 doesn't cause anything legible, a value of 9 on the other hand gives the best result as shown in the picture below. Higher or lower values will only show some line fragments but nothing even close to the expected content. Additionally, values over 17 cause the screen to stay blank. The `tile_height` only has an effect on the number of lines that are shown below the faint "Hello World!" writing. Values below 8 cause some empty space at the bottom of the display and values larger than 9 have no visible change. The picture below shows the result with a `tile_height` value of 9. 
To my surprise, switching the values of `pixel_width` and `pixel_height` (160 and 68) does not cause a noticeably different image on the screen.

[image]

For reference, this is the display info with the best result I managed to find:

```
static const u8x8_display_info_t u8x8_ls013b7dh03_128x128_display_info =
{
  /* chip_enable_level = */ 1,
  /* chip_disable_level = */ 0,
  /* post_chip_enable_wait_ns = */ 50,
  /* pre_chip_disable_wait_ns = */ 50,
  /* reset_pulse_width_ms = */ 1,
  /* post_reset_wait_ms = */ 6,
  /* sda_setup_time_ns = */ 227,	/* 227 nsec according to the datasheet */		
  /* sck_pulse_width_ns = */  255,	/* 450 nsec according to the datasheet */
  /* sck_clock_hz = */ 1000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* changed from 2 to 0 (https://github.com/olikraus/u8g2/issues/1771) */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 100,
  /* write_pulse_width_ns = */ 100,
  /* tile_width = */ 9,
  /* tile_height = */ 9,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 160,
  /* pixel_height = */ 68
};
```

At this point I've reached the end of what I can understand of the code to get the display working. Maybe I need to change something in other files that I don't know of yet, but at this point I don't know what else to try. Feel free to raise any suggestions or questions for me to look into.

olikraus, 2024-03-31:
added to 2.35 for later review

Phasikk, 2024-03-31:
> I tried recreating your experiment on Arduino Uno R4 with nice!view and the adafruit library, same code, same wiring(as you described it there). I couldn't make it work, unfortunately.
> 
> Turns out it was a skill issue. I recreated your experiment with other board ESP32 Dev Module and it worked first try. Thank you for providing a solution for this.

I'm glad my issue documentation was of some help, at least regarding the Adafruit library.

Off-topic regarding U8g2, but I did try my Adafruit test code with an UNO R4 WiFi (5V), the nice!view display (3.3V) and a level shifter in-between. (Arduino IDE 2.3.2, UNO R4 Boards core 1.1.0, Adafruit SHARP Memory Display library 1.1.3)
I can confirm this setup not working, since MOSI is stuck on HIGH no matter what pin I define for it.

Phasikk, 2024-03-31:
> added to 2.35 for later review

@olikraus: Sounds good! Let me know if I can help with any testing.

olikraus, 2024-08-31:
Thats a long pending topic... I have create the constructor U8G2_LS011B7DH03_160X68_F_4W_HW_SPI based on the input from above, however with some additional fixes.

I would be happy about some confirmation, that this works... :-)

The constructor is available with beta 2.35.21

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

Phasikk, 2024-09-01:
I can happily report that the display now works with U8g2. :)
[image]

This time I had to resort to an Adafruit QT Py SAMD21 board with the nice!view breakout for the Sharp LS011B7DH03 display. I used the most recent U8g2 version in PlatformIO and the new constructor with my previous test code.

| QT Py SAMD21 | nice!view LS011B7DH03 |
| --- | --- |
| GND | GND |
| 3.3V | VCC |
| A0 | CS |
| MOSI (pin 10)  | MOSI |
| SCK (pin 8) | SCK |

code:
```
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

U8G2_LS011B7DH03_160X68_F_4W_HW_SPI u8g2(U8G2_R0, A0, U8X8_PIN_NONE, U8X8_PIN_NONE);

void setup(void){
  u8g2.begin();
}
 void loop(void){
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB08_tr);
  u8g2.drawStr(0,10,"Hello World!");
  u8g2.sendBuffer();
  delay(1000);
}
```
platformio.ini:
```
[env:adafruit_qt_py_m0]
platform = atmelsam
board = adafruit_qt_py_m0
framework = arduino

lib_ignore = Adafruit TinyUSB Library
lib_deps =
	u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

I have also tested the U8g2 examples GraphicsTest, FPS and U8g2Logo with the full buffer mode and didn't notice any issues.

Thanks so much for adding support for this display! You have no clue how happy that makes me right now. :>
[image]
I realize once more that there are no obvious ways to donate/tip for this library. Is there any way I can support you and your amazing work?

olikraus, 2024-09-01:
Your lovely picture is donation enough, thank you for the feedback.

closed 2024-09-01 by olikraus