olikraus/u8g2 #2212: LCD screen glitching when powered

yees7, 2023-06-26:
I am using Arduino Nano. Every time I connect a cable from my computer to the Nano to power it, the LCD gets filled with these characters.  The LCD is supposed to display the current date and time, which update every second. The pixels that are supposed to update still seem to update when its filled with the characters. The only fix was to press the reset button on the Nano. Note that I am using serial communication through Python to get the time once (I update it every second after that), so resetting will reset the saved time data and I don't want that. When I replug it, it takes a few seconds for the LCD to show anything then it shows the random characters.  Normal: 
[image]
After unplugging and replugging: 
[image]
(I apologize for the bad image quality)
Here is my Arduino code:
```//-----------------libraries + definitions--------------//
//lcd libraries
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

// const int LEFT_BUTTON = 5, UP_BUTTON = 7, DOWN_BUTTON = 6, RIGHT_BUTTON = 4;
const int CLOCK = 12, DATA = 11, CS = 10;
U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, CLOCK, DATA, CS);

// time library (for calculating dates)
#include <TimeLib.h>

String dayAbr[] = {"Sun", "Mon", "Tue", "Wed", "Thu", "Fri", "Sat"};
//-----------------------------------//
unsigned long timeSeconds = 0;
unsigned long previousMillis = 0; //last time LED was updated

void drawTime() {
  // print time
  u8g2.setCursor(10, 50);
  int m = minute(timeSeconds);
  int s = second(timeSeconds);
  String timeStr = hourFormat12(timeSeconds) + String(":") + String(m<=9 ? "0" : "") + m + String(":") + String(s<=9 ? "0" : "") + s + (isAM(timeSeconds) ? " AM" : " PM");
  u8g2.print(timeStr);
  // print date
  u8g2.setCursor(10, 30);
  String dateStr = dayAbr[weekday(timeSeconds)-1] + String(" ") + day(timeSeconds) + String("/") + month(timeSeconds) + String("/") + year(timeSeconds);
  u8g2.print(dateStr);
}
void setup() {
  Serial.begin(115200);
  Serial.setTimeout(100);
  u8g2.begin();
  u8g2.enableUTF8Print();
}

void loop() {
  if (Serial.available()) {
    String inp = Serial.readStringUntil('\n');
    timeSeconds = inp.toInt();
  }

  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= 1000) {
    previousMillis = currentMillis-(currentMillis - previousMillis - 1000);
    timeSeconds++;
  }

  u8g2.firstPage();
  do {
    //--------------display-------------
    //font
    u8g2.setFont(u8g2_font_7x14_tf);
    //boundary
    drawTime();
  } while ( u8g2.nextPage() );
}```
How can I fix this problem without resetting the Nano? I did some research and [this issue](https://github.com/olikraus/u8g2/issues/853) had a similar issue and that the fix was to change the transfer speed, but their glitch was a bit different from mine. I am also not sure how to change the transfer speed. I suspect that this might be a hardware issue. Any reply is appreciated.

olikraus, 2023-06-26:
The ST7920 is a very sensitive display
- avoid long wires
- avoid protoboards
- use setBusClock to reduce the transfer speed (see u8g2 manual)
- use level shifter if you have a 3v3 board (the ST7920 expects 5V level)

yees7, 2023-06-27:
Hey! Thanks for the reply. Unfortunately, I am still getting the same results. 
I switched to my UNO instead because it had holes to hook up wires. The Nano had to be connected to a breadboard (which is the only board I had). Sadly, the only wires I had were hook-up wires so I could not control the length. I also put `u8g2.setBusClock(1000000);` before `u8g2.begin()` in setup(). I am not fully sure if this is the right value for the bus clock, I just used the lowest value it suggested for SPI.
It still glitches when unplugging and replugging. The screen seems to be a bit darker than it was with the Nano. The LCD did sometimes only glitch on a part of the screen, like only the top 10 rows of pixels. What do I do here?

olikraus, 2023-07-09:
hmmm....

NekoSamuraiDev, 2025-08-27:
did you find a solution to your problem?