olikraus/u8g2 #2213: Support for GP1294ai Display

StephiB, 2023-07-01:
Is it possible to add the GP1294AI VFD. I've tried to do it by myself but for some reason I can't get it to work. 
Datasheet:
 https://drive.google.com/file/d/1AG-xEty9XNawlXCaBtnWZ6uW1tDV3I1M/view?usp=drivesdk

Thanks in advance

olikraus, 2023-07-01:
Can you do a test with the other GP12xx constructors?
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#gp1287ai-256x50

What will you see? Which constructor works best? Could you sent an image?

StephiB, 2023-07-02:
No function at all, the initial setting of the Gp1247/87 seems to be totally different.
Gp1247ai :
https://drive.google.com/file/d/1AvN5UNpLIzMM8QzgwikIQiSCDImiOtRc/view?usp=drivesdk
Gp1287bi:
https://drive.google.com/file/d/1AvAbVk_QmEUBSvrYXAyUfUIyaGBQaEKs/view?usp=drivesdk
Gp1294ai:
https://drive.google.com/file/d/1AG-xEty9XNawlXCaBtnWZ6uW1tDV3I1M/view?usp=drivesdk

olikraus, 2023-07-09:
The main problem is: I am not the author of the GP12x7 displays... let's see...

olikraus, 2023-07-09:
I had a deeper look into the specification, but to me it looks like that the documentation is kind of incomplete, for example the description for the 0xe0 cmd seems to be missing. Second problem is, that I am not able to read far east languages. 

Maybe @izilzty can give better comments here...

izilzty, 2023-07-09:
Hi olikraus, it looks like GP1294 doesn't use 0x0e cmd. I think only needs to modify the init_seq and display_info of GP1287 to make it compatible with GP1294. I'll try to add support for GP1294, if I have time this week.

The datasheet uploaded by @StephiB is not from Futaba, it should be made by someone else, because the display is a custom model, so there is no public datasheet, but the commands in it can still be referred to.

olikraus, 2023-07-09:
> I'll try to add support for GP1294, if I have time this week.

Thanks, if you could do the init sequence suggestion, then I can do the remaining code changes.

medtech1, 2023-07-11:
I can't express  how much I'd love the GP1294 to be added to the list of supported displays.  I have two of them that sellers sent me in error, that I can't get anything to work with.  

The project I am trying is a VFD clock that works with the GP1287... but it does not work with the GP1294.   So I'm really glad to see this brought up.

medtech1, 2023-07-11:
btw, if it helps.. I found some documentation on the GP1294AI and had it translated using google:

https://redhat.systems/GP1294AI%20256x48%20VFD%20Display%20Module%20%28UNL-200AP-C%29%20-%20Jiali%20Chuang%20EDA%20Open%20Source%20Hardware%20Platform.htm

izilzty, 2023-07-12:
Hi medtech1, This is an Arduino library I created while testing the [PR#2222](https://github.com/olikraus/u8g2/pull/2222), which includes support for the GP1294AI.
[u8g2_arduino_2.35.2.zip](https://github.com/olikraus/u8g2/files/12022405/u8g2_arduino_2.35.2.zip)
You need to remove the current U8g2 library and install this zip library. But it should be noted that it is unofficial, when the official version supports GP1294AI, you can delete this library and reinstall the official version.

GP1294AI supports these constructors:
```
U8G2_GP1294AI_256X48_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
U8G2_GP1294AI_256X48_2_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
U8G2_GP1294AI_256X48_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);

U8G2_GP1294AI_256X48_1_2ND_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
U8G2_GP1294AI_256X48_2_2ND_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
U8G2_GP1294AI_256X48_F_2ND_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);

U8G2_GP1294AI_256X48_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/13, /* data=*/11, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
U8G2_GP1294AI_256X48_2_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/13, /* data=*/11, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/13, /* data=*/11, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);

U8X8_GP1294AI_256X48_4W_HW_SPI u8x8(/* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
U8X8_GP1294AI_256X48_2ND_4W_HW_SPI u8x8(/* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
U8X8_GP1294AI_256X48_4W_SW_SPI u8x8(/* clock=*/13, /* data=*/11, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);
```

medtech1, 2023-07-12:
Hi.. thank you!

In the code I was using this line for the GP1287

U8X8_GP1287AI_256X50_4W_SW_SPI u8x8(/* clock=*/ 0, /* data=*/ 2, /* cs=*/ 14, /* dc=*/ 4, /* reset=*/16);

I think this is the equiv for the GP1294:
U8X8_GP1294AI_256X48_4W_SW_SPI u8x8(/* clock=*/13, /* data=*/11, /* cs=*/10, /* dc=*/U8X8_PIN_NONE, /* reset=*/8);

I have to read the docs to see if the change in numbers is GPIO pin?

Thanks
Michelle

> Hi medtech1, This is an Arduino library I created while testing the PR#2222 <https://github.com/olikraus/u8g2/pull/2222>, which includes support for the GP1294AI.
> u8g2_arduino_2.35.2.zip [...]
`

izilzty, 2023-07-12:
Just like other constructor, you only need to change the pin number to the actual use. For example, you can use:
```
U8X8_GP1294AI_256X48_4W_SW_SPI u8x8(/* clock=*/ 0, /* data=*/ 2, /* cs=*/ 14, /* dc=*/ U8X8_PIN_NONE, /* reset=*/16);
 ```
GP1287, GP1247 and GP1294 did not use dc pin, using `U8X8_PIN_NONE` instead of pin number.

medtech1, 2023-07-12:
Hey you rock.. it worked.

 Thank you!!

> 
> 
> Just like other constructor, you only need to change the pin number to the actual use. For example, you can use:
> 
> `U8X8_GP1294AI_256X48_4W_SW_SPI u8x8(/* clock=*/ 0, /* data=*/ 2, /* cs=*/ 14, /* dc=*/ U8X8_PIN_NONE, /* reset=*/16);
`

medtech1, 2023-07-12:
Hey I sent you an email just to confirm your email address.. please let me know if it is correct.

izilzty, 2023-07-12:
I don't know how to find the email address on github. Anyway, my email address is izilzty@outlook.com

olikraus, 2023-07-12:
I am impressed... looks like all is solved.... wow... and there is also a PR. Impressive.

medtech1, 2023-07-12:
Yes, and thank you all.

olikraus, 2023-07-15:
Thanks again for the PR. Excellent work :-)
I have created beta release 2.35.3

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

4b616c69, 2023-07-15:
I'm currently having issues with this too and honestly I do not know how to fix this. 
I'm following this guide here: https://www.instructables.com/Simple-ESP8266-NTP-Clock-on-VFD-Display-VFD256x50-/
The VFD doesn't even power on. Is the pinout  for GP1294AI much different than the GP1287?

medtech1, 2023-07-15:
The pin-out on the NTP clock you are working with either the GP1287 or GP1294 is the same.  You just need to use:

ccoenen, 2023-09-21:
I did not see a pin mapping anywhere and duckduckgo brought me here, so I'm just mentioning it for future search engine enthusiasts:

Pins from the top to the bottom [taken from this datasheet](https://cdn.hackaday.io/files/1916648200152576/EPC-INBN0BV1294UD_SPEC.pdf):

```
CN7 1: FILMENT_EN (Filament enable, active HIGH)
CN7 2: SPI CLOCK
CN7 3: SPI Chip Select (active LOW / CSN)
CN7 4: SPI DATA (COPI, LSB First)
CN7 5: RESET (active LOW)
CN7 6: Ground

CN6 1: Ground
CN6 2: 5V output
CN6 3: AINL (audio in left channel)
CN6 4: AINR (audio in right channel)
CN6 5: LII_SW (Light sensor, GL5506 Pull down)
CN6 6: Ground
```

ccoenen, 2023-09-21:
RESET (grey in the instructable) was tied to D0, which did not work for me. I tied it to 3V3 (or probably any "HIGH" level) and it worked beautifully. Basically, all you need to connect is

* FILMENT_EN (any high level 3.3V/5V)
* SPI CLOCK (preferably to the HW SPI of your microcontroller)
* SPI Chip Select (oddly does not work by simply putting it on LOW level, needs to be connected to a pin as configured by the constructor)
* SPI DATA (preferably to the HW SPI of your microcontroller)
* RESET (tied to 3v3, as mentioned in the beginning)
* Ground (probably any of the three grounds) must be connected as well

Additionally, to have the VFD supply your microcontroller, you can use CN6 pin 2 (5V output).

olikraus, 2023-09-21:
> SPI Chip Select

The chip select usually will put the SPI communication state machine into a defined state. So yes, usually it is required.

Thanks for the updates and notes :-)

> duckduckgo brought me here

So, I hope you solved your problems with the help of u8g2

ccoenen, 2023-09-24:
For now, I'm just playing around with an interesting display. But without u8g2, this would have been much more work. So: Thanks, everyone! :-)

Niphoria, 2023-10-03:
Im also having issues getting this to run on the Pico W and at this point im kinda frustrated and im not sure if its a faulty module or me messing up constantly - anyway i hope someone can help me here and point out where i messed up

This is how i have wired the module up:

`    VFD         PICO                                                                                                        
Pin 1       Pin 20 (3.3V High)                                                                                           
Pin 2       Pin 4 (SCK)                                                                                                  
Pin 3       Pin 2 (CSn)                                                                                                  
Pin 4       Pin 5 (TX)                                                                                                   
Pin 5       Pin 20 (3.3V High)                                                                                           
Pin 6       Pin 38 (GND)                                                                                                 `

And this is the code im using:

```#include <iomanip>
#include <sstream>
#include <stdio.h>

//#include "hardware/dma.h"
//#include "hardware/i2c.h"
#include "hardware/spi.h"
//#include "pico/multicore.h"
#include "pico/stdlib.h"
#include "pico/time.h"
#include "pico/util/queue.h"

#include "pico/cyw43_arch.h"

#include <u8g2.h>
//#include "Libraries/u8g2functions.h"

u8g2_t u8g2;
int displayWidth = 256;
int displayHeight = 48;

// ADD THE INIT DISPLAY 
#define PIN_RST 15

#define SPI_PORT spi0
#define PIN_MISO 1
#define PIN_CS 2
#define PIN_SCK 4
#define PIN_MOSI 5
#define SPI_SPEED 4000 * 1000

// Not needed i guess
#define PIN_DC 20

uint8_t u8x8_byte_pico_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
  uint8_t *data;
  switch (msg) {
  case U8X8_MSG_BYTE_SEND:
    data = (uint8_t *)arg_ptr;
    spi_write_blocking(SPI_PORT, data, arg_int);
    break;
  case U8X8_MSG_BYTE_INIT:
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
    break;
  case U8X8_MSG_BYTE_SET_DC:
    u8x8_gpio_SetDC(u8x8, arg_int);
    break;
  case U8X8_MSG_BYTE_START_TRANSFER:
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
    u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO,u8x8->display_info->post_chip_enable_wait_ns, NULL);
    break;
  case U8X8_MSG_BYTE_END_TRANSFER:
    u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO,
                            u8x8->display_info->pre_chip_disable_wait_ns, NULL);
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
    break;
  default:
    return 0;
  }
  return 1;
}

uint8_t u8x8_gpio_and_delay_pico(u8x8_t *u8x8, uint8_t msg,uint8_t arg_int, void *arg_ptr) {
  switch (msg) {
  case U8X8_MSG_GPIO_AND_DELAY_INIT: 
    spi_init(SPI_PORT, SPI_SPEED);
    gpio_set_function(PIN_CS, GPIO_FUNC_SIO);
    gpio_set_function(PIN_SCK, GPIO_FUNC_SPI);
    gpio_set_function(PIN_MOSI, GPIO_FUNC_SPI);
    gpio_init(PIN_RST);
    gpio_init(PIN_DC);
    gpio_init(PIN_CS);
    gpio_set_dir(PIN_RST, GPIO_OUT);
    gpio_set_dir(PIN_DC, GPIO_OUT);
    gpio_set_dir(PIN_CS, GPIO_OUT);
    gpio_put(PIN_RST, 1);
    gpio_put(PIN_CS, 1);
    gpio_put(PIN_DC, 0);
    break;                  
  case U8X8_MSG_DELAY_NANO: // delay arg_int * 1 nano second
    sleep_us(arg_int); // 1000 times slower, though generally fine in practice given rp2040 has no `sleep_ns()`
    break;
  case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
    sleep_us(arg_int);
    break;
  case U8X8_MSG_DELAY_10MICRO: // delay arg_int * 10 micro seconds
    sleep_us(arg_int * 10);
    break;
  case U8X8_MSG_DELAY_MILLI: // delay arg_int * 1 milli second
    sleep_ms(arg_int);
    break;
  case U8X8_MSG_GPIO_CS: // CS (chip select) pin: Output level in arg_int
    gpio_put(PIN_CS, arg_int);
    break;
  case U8X8_MSG_GPIO_DC: // DC (data/cmd, A0, register select) pin: Output level
    gpio_put(PIN_DC, arg_int);
    break;
  case U8X8_MSG_GPIO_RESET: // Reset pin: Output level in arg_int
    gpio_put(PIN_RST, arg_int);  // printf("U8X8_MSG_GPIO_RESET %d\n", arg_int);
    break;
  default:
    u8x8_SetGPIOResult(u8x8, 1); // default return value
    break;
  }
  return 1;
}

void draw_display() {
	char hey[12] = "Hello world";
	u8g2_ClearBuffer(&u8g2);
        u8g2_ClearDisplay(&u8g2);
	u8g2_SetDrawColor(&u8g2, 1);
        u8g2_SetFont(&u8g2, u8g2_font_t0_11_te);
        u8g2_DrawStr(&u8g2, 10, 10, hey);
        u8g2_UpdateDisplay(&u8g2);
}

void display_sequence(){
  u8g2_Setup_gp1294ai_256x48_f(&u8g2, U8G2_R0, u8x8_byte_pico_hw_spi, u8x8_gpio_and_delay_pico);
  u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
  u8g2_SetPowerSave(&u8g2, 0); 
  draw_display();
}

int main()
{
    stdio_init_all;
    sleep_ms(1000);

    // Need to initialize the WIFI Module for the LED to work
    if (cyw43_arch_init() != 0)
    {
        printf("WIFI Module Failed!");
    }

    //multicore_launch_core1(display_core);
    

    while(true)
    {
        //display_sequence();
        printf("MEOW\n");
        sleep_ms(1000);
        cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, true);
        sleep_ms(1000);
        cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, false);
    }
};

izilzty, 2023-10-03:
Hi Niphoria, I have no experience with Pico, but I think it might be necessary to set the polarity and phase of the SPI. You can try adding the following code after `spi_init(SPI_PORT, SPI_SPEED);`

`spi_set_format(SPI_PORT, 8, SPI_CPOL_1, SPI_CPHA_1, SPI_MSB_FIRST); //SPI Mode 3`

Niphoria, 2023-10-03:
Hi izilzty, I applied your changes but sadly there is no sign of life on the VFD - is there any way i can test if the VFD module is even alive ? I have a power supply that can go up to 19V so i could apply it to the VFD display directly but i dont know if that would even be enough.

here are my code changes

```#include <iomanip>
#include <sstream>
#include <stdio.h>

//#include "hardware/dma.h"
//#include "hardware/i2c.h"
#include "hardware/spi.h"
//#include "pico/multicore.h"
#include "pico/stdlib.h"
#include "pico/time.h"
#include "pico/util/queue.h"

#include "pico/cyw43_arch.h"

#include <u8g2.h>
//#include "Libraries/u8g2functions.h"

u8g2_t u8g2;
int displayWidth = 256;
int displayHeight = 48;

// ADD THE INIT DISPLAY 
#define PIN_RST 22

#define SPI_PORT spi0
#define PIN_MISO 1
#define PIN_CS 2
#define PIN_SCK 4
#define PIN_MOSI 5
#define SPI_SPEED 4000 * 1000

// Not needed i guess
#define PIN_DC 20

//////////////////////////////////
// filament
#define FILAMENT 9
//////////////////////////////////

uint8_t u8x8_byte_pico_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
  uint8_t *data;
  switch (msg) {
  case U8X8_MSG_BYTE_SEND:
    data = (uint8_t *)arg_ptr;
    spi_write_blocking(SPI_PORT, data, arg_int);
    break;
  case U8X8_MSG_BYTE_INIT:
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
    break;
  case U8X8_MSG_BYTE_SET_DC:
    u8x8_gpio_SetDC(u8x8, arg_int);
    break;
  case U8X8_MSG_BYTE_START_TRANSFER:
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
    u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO,u8x8->display_info->post_chip_enable_wait_ns, NULL);
    break;
  case U8X8_MSG_BYTE_END_TRANSFER:
    u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO,
                            u8x8->display_info->pre_chip_disable_wait_ns, NULL);
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
    break;
  default:
    return 0;
  }
  return 1;
}

uint8_t u8x8_gpio_and_delay_pico(u8x8_t *u8x8, uint8_t msg,uint8_t arg_int, void *arg_ptr) {
  switch (msg) {
  case U8X8_MSG_GPIO_AND_DELAY_INIT: 
    spi_init(SPI_PORT, SPI_SPEED);
    //////////////////////////////////
    spi_set_format(SPI_PORT, 8, SPI_CPOL_1, SPI_CPHA_1, SPI_MSB_FIRST); //SPI Mode 3
    //////////////////////////////////
    gpio_set_function(PIN_CS, GPIO_FUNC_SIO);
    gpio_set_function(PIN_SCK, GPIO_FUNC_SPI);
    gpio_set_function(PIN_MOSI, GPIO_FUNC_SPI);
    gpio_init(PIN_RST);
    gpio_init(PIN_DC);
    gpio_init(PIN_CS);
    gpio_set_dir(PIN_RST, GPIO_OUT);
    gpio_set_dir(PIN_DC, GPIO_OUT);
    gpio_set_dir(PIN_CS, GPIO_OUT);
    gpio_put(PIN_RST, 1);
    gpio_put(PIN_CS, 1);
    gpio_put(PIN_DC, 0);
    break;                  
  case U8X8_MSG_DELAY_NANO: // delay arg_int * 1 nano second
    sleep_us(arg_int); // 1000 times slower, though generally fine in practice given rp2040 has no `sleep_ns()`
    break;
  case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
    sleep_us(arg_int);
    break;
  case U8X8_MSG_DELAY_10MICRO: // delay arg_int * 10 micro seconds
    sleep_us(arg_int * 10);
    break;
  case U8X8_MSG_DELAY_MILLI: // delay arg_int * 1 milli second
    sleep_ms(arg_int);
    break;
  case U8X8_MSG_GPIO_CS: // CS (chip select) pin: Output level in arg_int
    gpio_put(PIN_CS, arg_int);
    break;
  case U8X8_MSG_GPIO_DC: // DC (data/cmd, A0, register select) pin: Output level
    gpio_put(PIN_DC, arg_int);
    break;
  case U8X8_MSG_GPIO_RESET: // Reset pin: Output level in arg_int
    gpio_put(PIN_RST, arg_int);  // printf("U8X8_MSG_GPIO_RESET %d\n", arg_int);
    break;
  default:
    u8x8_SetGPIOResult(u8x8, 1); // default return value
    break;
  }
  return 1;
}

void draw_display() {
	char hey[12] = "Hello world";
	u8g2_ClearBuffer(&u8g2);
        u8g2_ClearDisplay(&u8g2);
	u8g2_SetDrawColor(&u8g2, 1);
        u8g2_SetFont(&u8g2, u8g2_font_t0_11_te);
        u8g2_DrawStr(&u8g2, 10, 10, hey);
        u8g2_UpdateDisplay(&u8g2);
}

void display_sequence(){
  u8g2_Setup_gp1294ai_256x48_f(&u8g2, U8G2_R0, u8x8_byte_pico_hw_spi, u8x8_gpio_and_delay_pico);
  u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
  u8g2_SetPowerSave(&u8g2, 0); 
  draw_display();
}

int main()
{
    stdio_init_all;
    sleep_ms(1000);

    // Need to initialize the WIFI Module for the LED to work
    if (cyw43_arch_init() != 0)
    {
        printf("WIFI Module Failed!");
    }

    //multicore_launch_core1(display_core);

    //////////////////////////////////
    gpio_init(FILAMENT);
    gpio_set_dir(FILAMENT, GPIO_OUT);
    gpio_put(FILAMENT, 1);
    //////////////////////////////////
    display_sequence();

    while(true)
    {
        
        printf("MEOW\n");
        sleep_ms(1000);
        cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, true);
        sleep_ms(1000);
        cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, false);
    }
};

izilzty, 2023-10-03:
After the program runs, if the filament in the display is dark red, it means that the power supply and FILMENT_EN pin are correct, and you may need to check the SPI wiring and program.

If you are not sure the display is alive, you can try running the example program using a 3.3V Arduino board, the example program on the Arduino board should be correct and you can concentrate on troubleshooting hardware issues.

power-dodge, 2023-10-14:
Helo,

I'm trying in vain to get this display to work with an ESP32 board. Use Arduino IDE to create programs. I think I'm connecting it incorrectly. I have the following board:

[https://mischianti.org/2021/02/17/doit-esp32-dev-kit-v1-high-resolution-pinout-and-specs/](url)

I haven't even managed to create a character yet. U8G2 is master installed

izilzty, 2023-10-14:
Hi, I have this ESP32 board too, maybe you should check the wiring again or try this example:

Pinout:

|ESP32|VFD Board|
|----|----|
|32|FILMENT_EN|
|18|CLOCK|
|25|CHIPSELECT|
|23|DATA|
|33|RESET|
|GND|GND|

Please connect the power supply of the VFD module and ESP32 by yourself, because our boards may be different.

Code:

```
/*

  U8g2Logo.ino

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

*/

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

/*
  U8g2lib Example Overview:
    Frame Buffer Examples: clearBuffer/sendBuffer. Fast, but may not work with all Arduino boards because of RAM consumption
    Page Buffer Examples: firstPage/nextPage. Less RAM usage, should work with all Arduino boards.
    U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.
    
  This is a frame buffer example.    
*/

// Please UNCOMMENT one of the contructor lines below
// U8g2 Contructor List (Frame Buffer)
// The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
// Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected

// U8G2_GP1294AI_256X48_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/25, /* dc=*/U8X8_PIN_NONE, /* reset=*/33);
U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/18, /* data=*/23, /* cs=*/25, /* dc=*/U8X8_PIN_NONE, /* reset=*/33);

// End of constructor list

//#define MINI_LOGO

void setup(void) {
  u8g2.begin();

  pinMode(32, OUTPUT);
  digitalWrite(32, HIGH); // Filament Enable
  // digitalWrite(32, LOW); // Filament Disable

  pinMode(2, OUTPUT);
  digitalWrite(2, HIGH); // Turn on LED on board
}

void drawLogo(void)
{
    u8g2.setFontMode(1);	// Transparent
#ifdef MINI_LOGO

    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb16_mf);
    u8g2.drawStr(0, 22, "U");
    
    u8g2.setFontDirection(1);
    u8g2.setFont(u8g2_font_inb19_mn);
    u8g2.drawStr(14,8,"8");
    
    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb16_mf);
    u8g2.drawStr(36,22,"g");
    u8g2.drawStr(48,22,"\xb2");
    
    u8g2.drawHLine(2, 25, 34);
    u8g2.drawHLine(3, 26, 34);
    u8g2.drawVLine(32, 22, 12);
    u8g2.drawVLine(33, 23, 12);
#else

    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb24_mf);
    u8g2.drawStr(0, 30, "U");
    
    u8g2.setFontDirection(1);
    u8g2.setFont(u8g2_font_inb30_mn);
    u8g2.drawStr(21,8,"8");
        
    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb24_mf);
    u8g2.drawStr(51,30,"g");
    u8g2.drawStr(67,30,"\xb2");
    
    u8g2.drawHLine(2, 35, 47);
    u8g2.drawHLine(3, 36, 47);
    u8g2.drawVLine(45, 32, 12);
    u8g2.drawVLine(46, 33, 12);
    
#endif
}

void drawURL(void)
{
#ifndef MINI_LOGO
  u8g2.setFont(u8g2_font_4x6_tr);
  if ( u8g2.getDisplayHeight() < 59 )
  {
    u8g2.drawStr(89,20,"github.com");
    u8g2.drawStr(73,29,"/olikraus/u8g2");
  }
  else
  {
    u8g2.drawStr(1,54,"github.com/olikraus/u8g2");
  }
#endif
}

void loop(void) {
  u8g2.clearBuffer();
  drawLogo();
  drawURL();
  u8g2.sendBuffer();
  digitalWrite(2, !digitalRead(2));
  delay(1000);
}
```

After download and reset the ESP32, the VFD should show the U8g2 logo and the LED on the ESP32 board will flash.

power-dodge, 2023-10-15:
Hello,

Thank you very much for the code and the pin assignment. It looks like I had a broken ESP32 and also a few others that no longer work. The built-in LED function is worth its weight in gold. Thank you for the help. 

What I wanted to ask was, have you ever played around with the brightness? Do you maybe know if it's full brightness in your code?

izilzty, 2023-10-15:
The brightness in the code is the default value, which is not very bright, but enough for you to see what is on the screen. You can use `u8g2.setContrast();` to adjust the brightness you want.

power-dodge, 2023-10-15:
Many thanks for the answer.

But can you explain why your code works but the examples from the Arduino IDE don't? I also added your line, but unfortunately to no avail:

`U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/18, /* data=*/23, /* cs=*/25, /* dc=*/U8X8_PIN_NONE, /* reset=*/33);`

`U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/18, /* data=*/23, /* cs=*/25, /* dc=*/U8X8_PIN_NONE, /* reset=*/33);`

Thanks

izilzty, 2023-10-15:
Line 11 has an incomplete block comment. You probably commented out all the code behind it.

power-dodge, 2023-10-15:
Yes, I only removed the commented one. Here is a small example from the Aurduino IDE, Helo World. Unfortunately, Helo World doesn't appear on the display, but your example works great. I'm not much of a code expert, so I don't really see what's going on...

`/*

  HelloWorld.ino

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

*/

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

/*
  U8g2lib Example Overview:
    Frame Buffer Examples: clearBuffer/sendBuffer. Fast, but may not work with all Arduino boards because of RAM consumption
    Page Buffer Examples: firstPage/nextPage. Less RAM usage, should work with all Arduino boards.
    U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.
    
*/

// Please UNCOMMENT one of the contructor lines below
// U8g2 Contructor List (Frame Buffer)
// The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
// Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected
//U8G2_NULL u8g2(U8G2_R0);	// null device, a 8x8 pixel display which does nothing
//U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 12, /* dc=*/ 4, /* reset=*/ 6);	// Arduboy (Production, Kickstarter Edition)
//U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_SSD1306_128X64_NONAME_F_3W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* reset=*/ 8);
//U8G2_SSD1306_128X64_NONAME_F_3W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* reset=*/ 8);
//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1306_128X64_ALT0_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // same as the NONAME variant, but may solve the "every 2nd line skipped" problem
//U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* reset=*/ 8);
//U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
//U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 16, /* data=*/ 17, /* reset=*/ U8X8_PIN_NONE);   // ESP32 Thing, pure SW emulated I2C
[... 289 lines cut ...]
//U8G2_S1D15300_LM6023_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_S1D15721_240X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_S1D15721_240X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_GU800_128X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_GU800_128X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_GU800_160X16_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_GU800_160X16_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/18, /* data=*/23, /* cs=*/25, /* dc=*/U8X8_PIN_NONE, /* reset=*/33);

// End of constructor list

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}
`

Thank you for Help

izilzty, 2023-10-15:
Maybe you can attach the .ino file directly, it looks like there are a lot of "*" symbols missing in the code

power-dodge, 2023-10-15:
Yes its works with Pin FILMENT_EN to 3.3. Im testing and learn about the display.

Thank you for you Help, you are the best man. Thank You

power-dodge, 2023-10-15:
Hello,

this is a ContrastTest (Filment on 3.3V)

`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/18, /* data=*/23, /* cs=*/25, /* dc=*/U8X8_PIN_NONE, /* reset=*/33);

// End of constructor list

void testContrast(uint8_t contrast)
{
  static char cstr[6];
  static char vstr[4];
  u8g2.setContrast(contrast);
  strcpy( cstr, u8x8_u8toa(contrast, 3));
  u8g2.firstPage();
  do {
    //u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.setFont(u8g2_font_courB24_tn);
    u8g2.drawStr(0,28, cstr);
    u8g2.drawBox(64, 0, 10, 64);
    u8g2.drawBox(76, 0, 2, 64);
    u8g2.drawBox(80, 0, 1, 64);
    u8g2.drawBox(82, 0, 1, 64);
    
    u8g2.drawBox(0, 32, 128, 10);
    u8g2.drawBox(0, 44, 128, 2);
    u8g2.drawBox(0, 48, 128, 1);
    u8g2.drawBox(0, 50, 128, 1);
    
  } while ( u8g2.nextPage() );
}

void setup(void) {

  /* U8g2 Project: SSD1306 Test Board */
  //pinMode(10, OUTPUT);
  //pinMode(9, OUTPUT);
  //digitalWrite(10, 0);
  //digitalWrite(9, 0);		

  /* U8g2 Project: T6963 Test Board */
  //pinMode(18, OUTPUT);
  //digitalWrite(18, 1);	
  
  u8g2.begin();  
}

void loop(void) {
  uint8_t i;
  i = 255;
  do
  {
    testContrast(255-i);
    delay(15);
    i-=5;
  } while( i != 0 );
  i = 255;
  delay(100);
  do
  {
    testContrast(i);
    delay(15);
    i-=5;
  } while( i != 0 );
  delay(100);
}
`

medtech1, 2023-10-15:
I'm  curious about increasing the brightness of the VFD, and it would affect the life of the VFD itself. I noticed in my tests that the GP1287BI runs much warmer than a GP1294AI (which the heat isn't very noticeable at all).  Also, how much would the power consumption increase.

The VFDs are very nice displays (and they work really well for me), but not sure if they would work very well in areas where there is bright sunlight.

izilzty, 2023-10-16:
Sunlight may accelerate phosphorus failure, but this takes a long time, which may not be as fast as the pixel burn-in speed.

slabua, 2023-10-16:
One of the purposes of the VFDs is that to be used in bright sunlight where other technologies fail to deliver.

slabua, 2023-10-17:
@power-dodge I think this display has 1024 levels of brightness.
Whether that has been mapped to 0-255 in the library I have not verified it yet.

izilzty, 2023-10-17:
The input value is multiplied by 4 for remapping.

https://github.com/olikraus/u8g2/blob/711b50444071df6cec18a91cc6d947bd02928dee/csrc/u8x8_d_gp1294ai.c#L129-L130

power-dodge, 2023-11-03:
Hello, 
how can you operate multiple displays with one ESP32. I also connected the CS to other pins and when I control them, everything works fine. Arduino IDE won't let me create multiple displays in the constructor. 

`U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/18, /* data=*/23, /* cs=*/26, /* dc=*/U8X8_PIN_NONE, /* reset=*/33);`

`u8g2_1  u8g21`

Dont work

Thank you very much

olikraus, 2023-11-03:
There should be an example in the u8g2 FAQ.

power-dodge, 2023-11-03:
Thanks, its works with _2, but its dont red more.

I have 2 Displays istal, the Display nr. 2 its only works. I think the problem is on void setup:

`U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2_1(U8G2_R0, /* clock=*/18, /* data=*/23, /* cs=*/25, /* dc=*/U8X8_PIN_NONE, /* reset=*/33); //Screen 1
U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2_2(U8G2_R0, /* clock=*/18, /* data=*/23, /* cs=*/26, /* dc=*/U8X8_PIN_NONE, /* reset=*/33); //Screen 2`

`void setup() {
 Wire.begin();
 u8g2_1.begin();
 u8g2_2.begin();

}

void loop() {
  u8g2_1.firstPage();
  do{
    u8g2_1.drawXBM(0,0,Honda_width,Honda_height,Honda_bits);
  }while(u8g2_1.nextPage());
  delay(1000);
  u8g2_2.firstPage();
  do{
    u8g2_2.drawXBM(0,0,Honda_width,Honda_height,Honda_bits);
  }while(u8g2_2.nextPage());
  delay(1000);

}`

olikraus, 2023-11-04:
See the FAQ:

https://github.com/olikraus/u8g2/blob/2fca97d2cc7377a7a71fd404d17826af81147143/doc/faq.txt#L293-L300

The reset line of the second object must be set to U8X8_PIN_NONE.

power-dodge, 2023-11-04:
Hello, 

thank you very much for the answer.  I tried something around and until then it didn't work.  The buffer thing will definitely be very problematic with 4 displays and only one ESP 32. I then decided to use a separate ESP32 for each display, which gives more flexibility etc. Today I tried to connect them with i2c, which wasn't that great either  .  
I decided on CAN, I only managed to get the CAN up and running today, one sends three others see the packets, but it's really soooo fast over CAN.  My plan is that different mods on other displays are switched via CAN, I could have solved it using Pinout High, but who knows what ideas you will come up with... Unfortunately the driver occupies TX and RX, there still needs to be a clock module and  on it and the decoder for K-Line, OBD1... As you can see, all of this would probably be a bit much for an ESP32.  In any case, thank you for the help.

polyfrog99, 2023-11-12:
Hello,

I'm getting this error when verifying/compiling code but I'm not sure why:

Compilation error: 'U8G2_GP1294AI_256X48_F_4W_SW_SPI' does not name a type; did you mean 'U8G2_GP1287AI_256X50_F_4W_SW_SPI'?

I've got u8g2 version 2.34.22 installed in arduino ide. Not sure why it recognises the GP1287AI constructor but not the GP1294AI ? Has anyone seen this before or knows what's going wrong?

I'm just using some simple test code:

`/*

HelloWorld.ino

Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

Copyright (c) 2016, olikraus@gmail.com
All rights reserved.

Redistribution and use in source and binary forms, with or without modification,
are permitted provided that the following conditions are met:

Redistributions of source code must retain the above copyright notice, this list
of conditions and the following disclaimer.

Redistributions in binary form must reproduce the above copyright notice, this
list of conditions and the following disclaimer in the documentation and/or other
materials provided with the distribution.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND
CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES,
INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF
MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR
CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT
NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT,
STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF
ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

*/

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

/*
U8g2lib Example Overview:
Frame Buffer Examples: clearBuffer/sendBuffer. Fast, but may not work with all Arduino boards because of RAM consumption
Page Buffer Examples: firstPage/nextPage. Less RAM usage, should work with all Arduino boards.
U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.

*/

// Please UNCOMMENT one of the contructor lines below
// U8g2 Contructor List (Frame Buffer)
// The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
// Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected

U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, / clock=/18, / data=/23, / cs=/25, / dc=/U8X8_PIN_NONE, / reset=*/33);

// End of constructor list

void setup(void) {
u8g2.begin();
}

void loop(void) {
u8g2.clearBuffer(); // clear the internal memory
u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
u8g2.drawStr(0,10,"Hello World!"); // write something to the internal memory
u8g2.sendBuffer(); // transfer internal memory to the display
delay(1000);
}
`

power-dodge, 2023-11-12:
Helo, 
Check the connection to display. Have you 3.3v from display to yours Chip? GND to?

polyfrog99, 2023-11-12:
I've got it wired up, but the error is stopping the code from compiling so it's not being uploaded to the Arduino. It's not recognising this constructor:

U8G2_GP1294AI_256X48_F_4W_SW_SPI

Not sure why though, as the library is installed and it recognises the GP1287AI constructor.

polyfrog99, 2023-11-17:
Sorry ignore me, I should have read this thread more thoroughly - I had the standard library and not the beta one, it compiles now :)

closed 2024-08-24 by olikraus