olikraus/u8g2 #2216: SSD1320 Scanline fix?

JanB97, 2023-07-04:
I am playing around with one of those two controller SSD1320 OLED displays.
As mentioned in #1351 I am seeing scanning lines in the image.
Is there a fix for this?
Has anyone played around with this since?

@olikraus thank you so much for the functions that allow to draw text and pictures over both controllers!

olikraus, 2023-07-04:
> As mentioned in https://github.com/olikraus/u8g2/issues/1351 I am seeing scanning lines in the image.

Maybe the timing differs for your OLED. Or the power supply is not good enough. Sometimes it helps to add a 1uF cap close to the display power supply lines. The timing is usually mentioned in the original datasheet. This timing then has to be programmed into the OLED after startup.

JanB97, 2023-07-05:
I unfortunately don't have a datasheet, but my display looks exactly like the one buildxyz-git in the issue 1351.
I'll try the cap and see if this helps.
I also noticed that its only really visible on larger lid up areas, that however is not really my usecase, so you can't notice it that much afterall.

JanB97, 2023-07-10:
I'll close this issue as it turns out this really is a non issue for me, thank your for the great library! :)
I am still hoping for the greyscale extension btw. ;)

closed 2023-07-10 by JanB97