olikraus/u8g2 #2218: Display not working properly (sometimes), SH1122 driver with STM32 microcontroller

songyw124, 2023-07-07:
Hello,

Thank you for your work regarding amazing u8g2 and u8x8 library.

I am currently using u8g2 library to display texts in OLED display which uses SH1122 driver, with the custom board with STM32 microcontroller.

However, I have found that texts are not displayed in display sometimes, as the first picture which I have attached. Strangely, when I do multiple resets, I can make that texts to be displayed properly on the display (randomly).

I am thinking that this might be due to reset problem, because once I can make texts to be displayed properly, all the remaining texts will be rendered properly, and vice versa.

I am currently using SPI 4-wire interface, and I am also attaching code (snippet which was edited so that unnecessary parts are removed) and datasheet of the display that I am using.

Thank you very much.

```
#include "main.h"
#include "../../Drivers/U8g2/u8g2.h"
#include "NanumSquareEb51_Set3.c"

SPI_HandleTypeDef hspi1;
SPI_HandleTypeDef hspi5;

static u8g2_t u8g2;

static uint8_t tx_buffer[100];

void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_SPI5_Init(void);

uint8_t u8x8_stm32_gpio_and_delay(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr) {
  switch (msg) {
  case U8X8_MSG_GPIO_AND_DELAY_INIT:
    HAL_Delay(1);
    break;
  case U8X8_MSG_DELAY_MILLI:
    HAL_Delay(arg_int);
    break;
  case U8X8_MSG_GPIO_DC:
    HAL_GPIO_WritePin(DISP_A0_GPIO_Port, DISP_A0_Pin, arg_int);
    break;
  case U8X8_MSG_GPIO_RESET:
    HAL_GPIO_WritePin(DISP_nRES_GPIO_Port, DISP_nRES_Pin, arg_int);
    break;
  }
  return 1;
}

uint8_t u8x8_byte_4wire_hw_spi(u8x8_t* u8x8, uint8_t msg, uint8_t arg_int, void* arg_ptr) {
  switch (msg) {
  case U8X8_MSG_BYTE_SEND:
    HAL_SPI_Transmit(&hspi5, (uint8_t*) arg_ptr, arg_int, 1000);
    break;
  case U8X8_MSG_BYTE_INIT:
    break;
  case U8X8_MSG_BYTE_SET_DC:
    HAL_GPIO_WritePin(DISP_A0_GPIO_Port, DISP_A0_Pin, arg_int);
    break;
  case U8X8_MSG_BYTE_START_TRANSFER:
    HAL_GPIO_WritePin(DISP_nCS_GPIO_Port, DISP_nCS_Pin, GPIO_PIN_RESET);
    __NOP();
    break;
  case U8X8_MSG_BYTE_END_TRANSFER:
    __NOP();
    HAL_GPIO_WritePin(DISP_nCS_GPIO_Port, DISP_nCS_Pin, GPIO_PIN_SET);
    break;
  default:
    return 0;
  }
  return 1;
}

int main(void)
{
  HAL_Init();
  SystemClock_Config();

  MX_GPIO_Init();
  MX_SPI5_Init();

  //----- LCD -----//
  u8g2_Setup_sh1122_256x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, u8x8_stm32_gpio_and_delay);
  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0);

  while (1)
  {
    u8g2_ClearBuffer(&u8g2);
    u8g2_SetMaxClipWindow(&u8g2);

    u8g2_SetFont(&u8g2, NanumSquareEb51);
    u8g2_DrawUTF8(&u8g2, 0, 50, "Hello");

    u8g2_SendBuffer(&u8g2);
    HAL_Delay(1000);

    //--------------------
    u8g2_ClearBuffer(&u8g2);
    u8g2_SetMaxClipWindow(&u8g2);

    u8g2_SetFont(&u8g2, NanumSquareEb51);
    u8g2_DrawUTF8(&u8g2, 0, 50, "HERO Lab");

    u8g2_SendBuffer(&u8g2);
    HAL_Delay(1000);
  }
}
```

[image]
[image]
[ZJY067_2.08_24P OLED.pdf](https://github.com/olikraus/u8g2/files/11975226/ZJY067_2.08_24P.OLED.pdf)

olikraus, 2023-07-09:
Not sure how I can help here...

songyw124, 2023-07-10:
I was wondering if there is some bug in the u8g2's display reset code, because I have followed the following initialisation process mentioned in the u8g2 wiki.

```
u8g2_Setup_sh1122_256x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, u8x8_stm32_gpio_and_delay);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);
```

Therefore, I guess there might be some problem during u8g2_Setup_sh1122_256x64_f(...) code, that might not reset all SH1122 driver display correctly. Is this kind of problem happening for the first time?

olikraus, 2023-07-10:
Indeed u8g2_InitDisplay will sent the init/setup code to SBM. 
So far I have never heard about any similar issue with this controller.

songyw124, 2023-07-12:
Thank you for your answer. I will try to debug what's happening then let you know.

olikraus, 2025-03-16:
It's a feature: Use the "f" constructor: `u8g2_Setup_sh1122_256x64_f`

See the warning in this section: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#buffer-size

TrevorCash, 2025-03-16:
Thanks that solved my issue - using the f version of the function works