olikraus/u8g2 #2225: ST7567S I2C New device

thinhdinh25, 2023-07-16:
I can use this lcd with U8G2_ST7567_HEM6432_F_HW_I2C u8g2 but cannot use all lcd with res 128x64. Can u expand it please

olikraus, 2023-07-16:
For a 128x64 ST7567 try

U8G2_ST7567_JLX12864, U8G2_ST7567_ENH_DG128064, U8G2_ST7567_ENH_DG128064 or U8G2_ST7567_OS12864

See: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

thinhdinh25, 2023-07-16:
> For a 128x64 ST7567 try
> 
> U8G2_ST7567_JLX12864, U8G2_ST7567_ENH_DG128064, U8G2_ST7567_ENH_DG128064 or U8G2_ST7567_OS12864
> 
> See: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

i tried all but none worked. I thought they were for spi but not for i2c. Someone got it working here: https://github.com/olikraus/u8g2/discussions/2134. I tried to make changes but did not work