olikraus/u8g2 #2226: Doesn't work with ESP32-C3

berkut0, 2023-07-17:
I tried almost every combination on my Wemos C3 mini and none of the options worked. I've tried putting an I2C address, changing SDA and SCL, combining settings, but all to no avail. There is some kind of bug, as there are libraries that work with the same device fine

For example, that works: https://github.com/ThingPulse/esp8266-oled-ssd1306

[image]

olikraus, 2023-07-18:
Which constructor did you use

berkut0, 2023-07-18:
I tried all combinations of 128x64 I2C with different pin combinations.
<details> <summary>Screenshot</summary>
[image]
</details>

berkut0, 2023-07-23:
I think it's related
https://github.com/olikraus/u8g2/issues/1720
I went the same way, tried setting I2C address, manipulating pins, wiring

olikraus, 2023-07-23:
For u8g2 the i2c address has to be multiplied by 2 (see reference manual).

berkut0, 2023-07-23:
You're amazing!
Thank you so much!

I'm not sure where to find this in the reference manual (I've searched and haven't found it). But it works!

Doubled I2C address works!

olikraus, 2023-07-23:
Here: https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

See the argument description. There are also several issues related to this question.

berkut0, 2023-07-23:
Should it always be doubled, no matter if it's C3 or another chip?

olikraus, 2023-07-24:
Yes.

berkut0, 2023-07-25:
Thank you!
I don't want to be intrusive, but I feel like there's a certain heterogeneity. Let's say it's necessary. Can we make this more obvious, at least in examples? I have never worked with an i2c device where the address had to be doubled. I'm not the kind of experienced developer who can say that it's not a common thing, but it really feels like a very specific detail

schuppeste, 2023-11-08:
Maybe, the Bus clock, in Standard in arduino esp32 it is fixed to 1MHz.. the ssd1306 has a Maximum specification on 400Khz. I cant find the u8G2 init to 400khz in source. 

You can try u8g2.setBusClock(400000); before u8G2.begin()?

My fail?  I dont habe a C3 to Test IT.

olikraus, 2023-11-08:
> Thank you! I don't want to be intrusive, but I feel like there's a certain heterogeneity. Let's say it's necessary. Can we make this more obvious, at least in examples? I have never worked with an [...]

Hmm, I guess I missed to answer this.
This topic has been discussed very often meanwhile here in this project. I agree, that this "double" value of the i2c address is unusual and maybe wrong. However there are examples, where the "doubled" i2c address is used by the display vendor.

See here for further info: https://github.com/olikraus/u8g2/issues/821