olikraus/u8g2 #2228: Bug in drawUTF8X2 when x < 0

CNawalol, 2023-07-18:
https://github.com/olikraus/u8g2/assets/61059886/e1428c93-6e29-40de-85b0-1b57445786b1

code:
```
#include <Arduino.h>
#include "U8g2lib.h"

U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, 5,  4, 22);

void setup() {
    u8g2.begin();
    u8g2.enableUTF8Print();
    u8g2.setFont(u8g2_font_wqy16_t_gb2312);
}

int offset = u8g2.getWidth();
void loop() {
    u8g2.clearBuffer();
    u8g2.drawUTF8(0,12,std::to_string(offset).c_str());
    u8g2.drawUTF8(offset -= 1,30,"123456789QWERTYUIOPASDFGHJKLZXCVBNM123456789QWERTYUIOPASDFGHJKLZXCVBNM");
    u8g2.drawUTF8X2(offset -= 1,60,"123456789QWERTYUIOPASDFGHJKLZXCVBNM123456789QWERTYUIOPASDFGHJKLZXCVBNM");
    u8g2.sendBuffer();
    delay(30);
}
```

I tried drawStrX2 but the problem is still the same

Thank you very much.

olikraus, 2023-07-23:
Thanks for pointing out this issue. It is fixed in beta 2.35.4:

#
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

CNawalol, 2023-07-23:
Thank you, it works well.

closed 2023-07-23 by CNawalol