olikraus/u8g2 #2229: Screen corruption ESP32 + 2.4'' OLED

mckowa, 2023-07-19:
I am using the Espressif ESP32-DevKitC V4 development board with a 2.4-inch OLED display. In my project, I am utilizing the U8g2 library to control the display. However, I have encountered screen corruption and artifacts when attempting to draw lines on the OLED screen. Notably, there are no issues when drawing text ("hello world") on the display.

```c++
#include <Arduino.h>
#include <U8g2lib.h>

#define SDA_PIN 21
#define SCL_PIN 22

U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R1, /* clock=*/ SCL_PIN, /* data=*/ SDA_PIN, /* reset=*/ U8X8_PIN_NONE);

void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.clearBuffer();
  u8g2.drawHLine(0, 64, 64);
  u8g2.drawLine(30, 10, 30, 100);
  u8g2.sendBuffer();

  delay(1000);
}
```

[image]

olikraus, 2023-07-23:
Maybe you use the wrong constructor. 2.4 inch OLEDs can't be driven by a SSD1306 controller. It might be a SSD1309 or SSD1325. So, I mean, the constructor might not fit to the display.

mckowa, 2023-07-23:
I've used this screen with this constructor in my Arduino-based project with no problems.

olikraus, 2023-07-23:
I have tested this and for me everything works as expected.