olikraus/u8g2 #2233: ST7586S JLX384160

mspudi, 2023-07-31:
PrintHelloWorld.ino 
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
//#endif
//#ifdef U8X8_HAVE_HW_I2C
//#include <Wire.h>
//#endif

U8G2_ST7586S_JLX384160_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/4, /* data=*/ 18, /* cs=*/ 5, /* dc=*/ 17, /* reset=*/ 16);
U8G2_ST7586S_JLX384160_1_4W_SW_SPI  

Test unsuccessful

I also successfully tested ST7565 12861 on the motherboard
ESP32  TTGO T-BEAM
arduino ide 2.1 u8g2 2.3422      \u8g2 2.3544

mspudi, 2023-08-01:
Activate ESP32  16bit   Or activate u8g2 16bit?   
s there any way to activate u8g2 16bit

closed 2023-08-02 by mspudi

olikraus, 2023-08-04:
> Activate ESP32 16bit Or activate u8g2 16bit? s there any way to activate u8g2 16bit

yes, see FAQ: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L173