olikraus/u8g2 #2237: MUI_FORM renders on half of the screen

shikshik, 2023-08-05:
Using U8G2_SH1122_256X64_F_4W_HW_SPI, everything works fine with the normal rendering. When using MUI_FORM with x,y positions, it only renders on half the width of the display.

16BIT is already enabled for samd21 (and it also renders on a full width when not using the MUI).

Any suggestions?

olikraus, 2023-08-05:
hmm... did I made a mistake here?
https://github.com/olikraus/u8g2/blob/08f70e80b44aa0673d6fde9901566ad3360923f6/csrc/mui_u8g2.c#L212-L217

Probably line 215 should be "return ui->x * 2"

MUI coordinates are only from 0..255, so the idea here is to change this to 0..127 for large displays.
Can you change line 215 in your local copy of MUI? 
After this, coordinates 0..127 can be used to write to the complete display.

shikshik, 2023-08-05:
Hey, thank you for your quick reply.
Yes, your quick fix did work.
The 0-127 grid system is very hard to manage when using the "MUI_FORM" with XY positions.
For now, I think I will skip MUI and use the basic methods.

Thank you for a great library!

shikshik, 2023-08-09:
I made a PR with the fix :)

shikshik, 2023-08-10:
Can you please update the platformIO package?

olikraus, 2023-08-13:
I am not the owner of the platformIO package... no Idea what I could do for this.

olikraus, 2024-08-27:
The PR is already merged, so I think we can close this issue...

closed 2024-08-27 by olikraus