olikraus/u8g2 #2238: First Arduino example code on wiki doesn't compile

nolanhergert, 2023-08-05:
From here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#introduction

/* Constructor */
U8G2_UC1701_DOGS102_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

Fails with 
src\main.cpp:6:1: error: 'U8G2_UC1701_DOGS102_1_4W_SW_SPI' does not name a type
 U8G2_UC1701_DOGS102_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
 ^

It looks like you can just replace the two instances of:
U8G2_UC1701_DOGS102_1_4W_SW_SPI 
with
U8G2_UC1701_EA_DOGS102_1_4W_SW_SPI 

and it just works.

olikraus, 2023-08-05:
Oh, so it's a documentation error?

nolanhergert, 2023-08-05:
Yes, I believe so. It's not tested anywhere specifically so it wasn't updated when the code changed. Not sure why the original constructor was removed and EA added.

olikraus, 2023-08-05:
Hmm Indeed a documentation issue, The correct constructor is listed here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#uc1701-ea_dogs102

Nice finding. Thanks for reporting this.

olikraus, 2024-08-24:
finally fixed...

closed 2024-08-24 by olikraus