olikraus/u8g2 #2241: ST7920 128x32 Support

ITstreet1, 2023-08-11:
As the title says, does this lib support the ST7920 LCD display in a resolution of 128x32? It has a 7-pin header.
https://www.aliexpress.com/item/1005004110848636.html

olikraus, 2023-08-11:
Can you check, whether the 
U8G2_ST7920_128X64_1_SW_SPI or U8G2_ST7920_128X64_1_HW_SPI
constructor works?

ITstreet1, 2023-08-11:
Ok,
I used this constructor
U8G2_ST7920_128X64_1_SW_SPI and Uno. 
Connection as on
U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ 8);
I tried HelloWorld and ContrastTest sketches but got nothing.

[image]

[image]

However, one of the buyers has written that the pinout on the display is NOT good. At the bottom, there are marks 1 and 7 on the sides of the header.
How to check it out? Any suggestions?

ITstreet1, 2023-08-11:
[image]
On my display, I have this header labeled with 1 at the bottom and 7 at the top.

Firstly, I connected it like this:
7 - 5V
6 - pin 8
5 - pin 13
4 - pin 11
3 - pin 10
2 - 5V
1 - GND

Later, when I read the comments, I used a multimeter to check the pattern from the header, and those two 7-tail pads. The pattern is not the same. 
The pattern is like this:
Header - pads
1 - 1
2 - 5
3 - 6
4 - 7
5 - 2
6 - 3
7 - 4

Not sure did I burned the display while the first connection, hope not.

Can you suggest to me what to try?

ITstreet1, 2023-08-11:
Further investigation I found that the BL LED is ok. And the pinout of the table from the seller's page is a GARBAGE.

The GND is on pin 5 of the header, and the BL (3.3V) is on pin 1 of the header. The backlight LED is working on 3.3V.
Two done, five to go.

xygax, 2023-11-20:
Guys 
I use these display's. they are labels ZXM12832C1
this constructor works 
#include <U8g2lib.h>
U8G2_ST7920_144X32_F_SW_SPI u8g2(U8G2_R2, SCLK ,  MOSI, CS, RST);

pinouts are as follows (square pin is pin 1)
1. BL +VE 5V
2. /CS
3. MOSI
4. Clock
5. ground (0V)
6. VDD (3.3V)
7. /Reset

xygax, 2023-11-20:
Oliver This is the one i use it works (just a bit weird when the screen is upside down needs an offset)
> U8G2_ST7920_144X32_F_SW_SPI u8g2(U8G2_R2, SCLK , MOSI, CS, RST);

ITstreet1, 2023-11-20:
I will try and let you know, as soon as I get back home. This is nice news.

olikraus, 2023-11-23:
I think anothe rending issue is to create the correct constructor for this display... I just added this to the latest milestone so that I don't miss this.

xygax, 2023-11-23:
Thanks. . I have also lookd another issue cant remember the number on this display and corrected the electricals as the chinese data is way wrong ...SSent from my Galaxy

ITstreet1, 2023-11-23:
@xygax 

I used the constructor you provided and got something

[image]

Obviously not what I expected.

This is a Hello World example
U8G2_ST7920_144X32_F_SW_SPI u8g2(U8G2_R2, 13 , 11, 10, 8);

xygax, 2023-11-23:
Two things... One ..... set the rotation to R0 and it will work but upside down Two. Set the origin to 20.x and that will be the difference between 128 the display size and 144 the constructor size SteveSent from my Galaxy

ITstreet1, 2023-11-23:
Origin to... what?

xygax, 2023-11-23:
look at the constructor ...... it has R2..... that is use the screen upside down make that R0 you will get your example code working...

or leave the constructor and display my clock

void LCD_Clock(void) {

  u8g2.clearBuffer();                                                 // clear the internal memory
  u8g2.setFont (u8g2_font_maniac_te);                   // 1 lines 22 point
  //u8g2.setFont(u8g2_font_ncenB08_tr);               // choose a suitable font
  u8g2.setCursor(19, 28);                                        //was 4.28 but moved due to constructor size 144
  u8g2.print(Timeload);                                          // write something to the internal memory timeload is my string xx:xx:xx
  u8g2.sendBuffer();                                                // transfer internal memory to the display
  }

ITstreet1, 2023-11-23:
I know about the rotation, but thought there is a tweak to fix the screen resolution difference.

xygax, 2023-11-23:
Not yet.. its on Oliver's list so i just correct it manually by the difference in the actual screen size and the constructor size

olikraus, 2023-11-24:
ah... yeah... too less time for u8g2 at the moment :-p

xygax, 2023-12-03:
> Hi All
> I have added the 128x32 constructor. Not sure about the left side offset...
> Constructor prefix is U8G2_ST7920_128X32_ in the new beta release 2.35.8
> Maybe someone can execute FlipMode.ino example (See example menu in Arduino IDE or here). The flip mode test example will cycle through eight modes. In all modes there must be a solid frame next to the border of the display.
> You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
> Arduino IDE:
>
> Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
> Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).
>
> PlatformIO:
> platformio.ini (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include
> lib_deps =
>   u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip

I have a unit on my desk will do it today.SSent from my Galaxy

olikraus, 2023-12-03:
Seems to be fine according to @xygax

olikraus, 2024-09-02:
so... closing this for now...

closed 2024-09-02 by olikraus