olikraus/u8g2 #2245: Overflow of main.elf file

ArthurFleischman, 2023-08-15:
while i was creating my custom multistage makefile(one makefile for lib other for main program) i found myself facing this error

## error
```
/opt/homebrew/opt/avr-binutils/bin/avr-ld: main.elf section `.text' will not fit in region `text'
/opt/homebrew/opt/avr-binutils/bin/avr-ld: address 0x830a51 of main.elf section `.bss' is not within region `data'
/opt/homebrew/opt/avr-binutils/bin/avr-ld: address 0x830a51 of main.elf section `.bss' is not within region `data'
/opt/homebrew/opt/avr-binutils/bin/avr-ld: region `text' overflowed by 6784724 bytes
collect2: error: ld returned 1 exit status
make: *** [main.elf] Error 1
```
## folder structure

```
├── CODEOWNERS
├── Makefile
├── Makefile1
├── README.md
├── adc
│   ├── Makefile
│   ├── adc.a
│   ├── adc.c
│   ├── adc.h
│   └── obj
├── compilation.log
├── main.c
├── obj
│   └── main.o
└── u8g2
    ├── Makefile
    ├── lib
    ├── obj
    ├── src
    └── u8g2.a
```
---
## `u8g2/Makefile` steps
. compile all .c to .o
. compress all .o in .a

## `./Makefile` steps
. botstrap u8g2 and other libs build
. builds main.c to main.o
. builds main.elf linking with -lm the .a

ArthurFleischman, 2023-08-15:
[compilation.log](https://github.com/olikraus/u8g2/files/12353630/compilation.log)
file with the build log

ArthurFleischman, 2023-08-15:
## root makefile
```
TARGET = main
SRCDIR = .
OBJDIR = ./obj
PROGRAMMER = usbasp
F_CPU = 16000000UL
# List of library directories
LIB_DIRS = u8g2

# List all the source files for the main program
SOURCES = $(wildcard $(SRCDIR)/*.c)
# Generate the object file names for the main program
OBJECTS = $(patsubst $(SRCDIR)/%.c,$(OBJDIR)/%.o,$(SOURCES))

OBJ = $(foreach dir,$(LIB_DIRS),$(wildcard $(dir)/obj/*.o))
ARCHIVES = $(foreach dir,$(LIB_DIRS),$(wildcard $(dir)/*.a))
LD_ARCHIVES = $(foreach dir,$(ARCHIVES),-lm $(dir))

# Compiler options for the main program
CC = avr-gcc
MCU = atmega328p
CFLAGS = \
	-mmcu=$(MCU) \
	-Os \
	-Wall \
	-Werror \
	-ffunction-sections \
	-fdata-sections \
	-Iu8g2/src/ \
	-Iu8g2/lib/ \
	-DAVR_USE_HW_I2C

# Default target for the main program
all: .PHONY $(TARGET).elf

# Rule to build the main program
$(TARGET).elf: $(OBJECTS)
	@echo building main.elf file
	$(CC) $(CFLAGS) $(OBJ)  $<  -o $@

# Rule to build object files for the main program
$(OBJDIR)/%.o: $(SRCDIR)/%.c
	@echo "building main"
	$(CC) $(CFLAGS) -c $< -o $@

# Rule to create hex file from ELF file
$(TARGET).hex: $(TARGET).elf
	@echo building main.hex
	avr-objcopy -O ihex -R .eeprom $< $@

# Rule to program the microcontroller using avrdude
program: $(TARGET).hex
	avrdude -c $(PROGRAMMER) -p $(MCU) -b 115200 -U flash:w:$(TARGET).hex

# Autobuild library Makefiles
.PHONY: $(LIB_DIRS)
$(LIB_DIRS):
	@echo building deps
	$(MAKE) -C $@

size.%:
	$(SIZE) --mcu=$(MCU) --format=avr $@

# Clean rule for the main program and libraries
clean: clean_libs
	rm -rf $(OBJDIR)/*.o $(TARGET).elf $(TARGET).hex

# Clean rule for libraries
clean_libs:
	@for dir in $(LIB_DIRS); do \
		$(MAKE) -C $$dir clean; \
	done
```
---
## u8g2 makefile
```

# Define the target and source files for lib1
TARGET = u8g2.a
SRC = lib/avr-hw-i2c lib src
OBJ = obj
F_CPU = 16000000UL
# List all the source files for lib1
SOURCES := $(foreach dir,$(SRC),$(wildcard $(dir)/*.c))

# Generate the object file names for u8g2
OBJECTS := $(patsubst %.c,$(OBJ)/%.o,$(notdir $(SOURCES)))
# Compiler options for lib1
CC = avr-gcc
CFLAGS =  -Os -mmcu=atmega328p \
	-Wall \
	-DF_CPU=$(F_CPU) \
	-DAVR_USE_HW_I2C \
	-Werror \
	-ffunction-sections \
	-fdata-sections \
	-Iu8g2/src/ \
	-Iu8g2/lib/ \
	-DAVR_USE_HW_I2C \
	-Wl,--gc-sections

# Linker options for lib1
AR = avr-ar
ARFLAGS = rcs

# Default target for lib1
all: $(TARGET)
# all: 
# 	@echo $(OBJECTS)
# 	@echo $(SOURCES)

# Rule to build the library for lib1
$(TARGET): $(OBJECTS)
	$(AR) $(ARFLAGS) $@ $(OBJECTS)

# Rule to build object files for lib1
$(OBJ)/%.o: */%.c
	$(CC) $(CFLAGS) -c $< -o $@

$(OBJ)/%.o: */*/%.c
	$(CC) $(CFLAGS) -c $< -o $@

# Clean rule for lib1
clean:
	rm -rf $(OBJ)/*.o $(TARGET)

```

olikraus, 2023-08-15:
The "-Wl,--gc-sections" argument must be passed to the linking step, however it is passed to compilation step in you files.
In other words, "-Wl,--gc-sections" is not required in the current file but is missing in the other file.

To be on the safe side, you could keep all three flags together "-ffunction-sections -fdata-sections -Wl,--gc-sections" and pass all three of them during compilation and linking. For you this means: Add "-Wl,--gc-sections" to the CFLAGS for the first file.

ArthurFleischman, 2023-08-15:
fixed with https://github.com/olikraus/u8g2/issues/618
all ok

closed 2023-08-15 by ArthurFleischman