olikraus/u8g2 #2246: Clear particular pixels and not entire display

himanshushah05, 2023-08-16:
How can we clear just particular pixles on the screen and not the entire display using clearDisplay() ?

v1993, 2023-08-16:
Wouldn't this be just `setDrawColor(0)` followed by `drawBox` with appropriate values? Don't forget to call `setDrawColor()` afterwards to restore normal drawing mode for other operations.

himanshushah05, 2023-08-23:
> Wouldn't this be just `setDrawColor(0)` followed by `drawBox` with appropriate values? Don't forget to call `setDrawColor()` afterwards to restore normal drawing mode for other operations.
> 
> Speaking of which - `clearDisplay()` docs note that it is usually not needed as `sendBuffer`/`nextPage` already clear the buffer. So you shouldn't rely on image persisting between frames, it would seem.

Oh yeah, worked perfectly fine
Thank you

closed 2023-08-23 by himanshushah05