olikraus/u8g2 #2250: Just glitches on ESP32 + SSD1322U 3.11" 256x64 OLED Display

hipur, 2023-08-24:
I've been trying to make things work for the past 12 hours with no luck, trying a simple sketch as follows:

`#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 0, /* dc=*/ 2, /* reset=*/15);	

void setup() {
  Serial.begin(115200);                                 
  Serial.println("Booting..");
  u8g2.begin();
}

void loop() {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,20,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}`

All I get are some screen flashes and glitches, like an old TV with a broken tube and without antenna signal.
I tried with ESP32 Dev Kit V1 and ESP32 DevKitC V4. Tried all possible GPIOs and nothing works.

My library came with U8G2_16BIT already enabled in u8g2.h and using SW SPI gives me no image at all.

This is my display:
https://www.aliexpress.com/item/32949282762.html

I've resoldered it in order to enable 4W_SPI as printed on the board.
I am clueless, so any help is appreciated.

olikraus, 2023-08-24:
Using SW_SPI does not require any resoldering. Just assign the pin numbers to the constructors, which are used for SPI Data and SPI clock.

You should also crosscheck, whether the SPI interface is configured as described on the backside of the PCB.
You could also post a picture of the back side of the PCB of the OLED so that we can crosscheck this together. I mean, you might have ordered SPI, but you could have received the same OLED configured as I2C or something else.

A picture of your complete setup might be also good to see if an other soldering issues are there.

hipur, 2023-08-24:
Sure, here are the pics for the OLED pinout, 4W SPI correctly selected and connections

[image]
[image]
[image]

I think I'm not missing anything, but you never know...

olikraus, 2023-08-24:
PCB looks good. The next step should be a mapping table, between display and ESP32 pins, like this:

| Display Pin | ESP Pin | Wire Color | Note |
|---|---|---|---|
| 1 | ? | Black | VSS, GND |
| 2 | ? | Gray? | 5V |
| 4 | 19?  | Green | SPI Clock |
| 5 | ? | Orange  | SPI Data |
| 14 | | | |
| 15 | | | |
| 16 | | | |

Numbers are a little bit difficult to read, but I think you can complete the table and post it here.
Then once this is done, we can take the numbers of the second column (ESP Pin) and put them into the SW_SPI constructor as arguments.

hipur, 2023-08-24:
Okay, I'm just not gonna use colors because I'm colorblind, haha.

| Display Pin | ESP Pin | Note |
| --- | --- | --- | 
| 1 | None | External GND |
| 2 | None | External 4.2V |
| 4 | 18 | SPI Clock |
| 5 | 23 | SPI Data |
| 14 | 2 | D/C |
| 15 | 15 | RST |
| 16 | 0 | CS/SS |

If I use these pins on the SW_SPI constructor I get a black screen, not even the glitches show up anymore.

olikraus, 2023-08-24:
the constructor should be:
`U8G2_SSD1322_NHD_256X64_1_4W_SW_SPI(rotation, clock, data, cs, dc [, reset]) `
So in your case this is
`U8G2_SSD1322_NHD_256X64_1_4W_SW_SPI(U8G2_R0, 18, 23, 0, 2,15) `
correct?
If this still doesn't work, please run the following constructor:
`U8G2_SSD1322_240X128_1_4W_SW_SPI(U8G2_R0, 18, 23, 0, 2,15) `
together with the FlipMode.ino example. Will you see something?
The FlipMode example is available in the Arduino IDE or from here: https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/FlipMode

hipur, 2023-08-24:
Both F and 1 constructors for both resolutions resulted in this being displayed:
[image]

After 1 min this image goes away and screen stays black. At least SW SPI is displaying something now.

olikraus, 2023-08-25:
Did  you use the flipmode example?
Did you try the U8G2_SSD1322_240X128_1_4W_SW_SPI  constructor?

achillhasler, 2023-08-25:
I recommend to connect the ground pins of display and esp board.

Fishbone69, 2023-08-29:
Did you ever get this resolved?  Your screenshot looks very similar to mine.  I'll post when I get back from work.

hipur, 2023-08-30:
> Did you use the flipmode example? Did you try the U8G2_SSD1322_240X128_1_4W_SW_SPI constructor? Which code did you use to generate the picture you have uploaded?

Sorry for the delay. I tried both constructors with the flipmode example. The picture was using it.

> Did you ever get this resolved? Your screenshot looks very similar to mine.
Funny enough I went to bed and just left the board with display on my desk. Next morning when I tried to boot up it just worked.

Transferred the code to another ESP32 unit and it was working again. I honestly could never figure out what actually happened or how I fixed it.

closed 2023-08-30 by hipur