olikraus/u8g2 #2251: Recent Port of Code from ESP32-C3 to ESP32-S3 Has Issues

Fishbone69, 2023-08-27:
Ollie:
     I took that digital watch code I showed you and tried to port it to WEMOS Lolin ESP32-S3 mini controller and the display was a garbled mess.  So I Started a new project and only ran the GraphicsTest.ino that comes with U8g2.  It was also a garbled mess.  I am using the following constructor:
U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* DC=*/ 13, /* reset=*/ 43);

I have checked to make sure all of the pins are correct - 
CLK=12
MOSI=11
MISO=DC=13 (see constructor)
CS=10

Have you had any other issues with this board or with esp32-s3 in general?  I checked the pins.arduino.h file as well and the default pins seem to match.  

Any help is appreciated.

Fish

Fishbone69, 2023-08-27:
********************************
Follow-up
**********************************
I tried a different constructor for a "similar" display but one that specifies the clock and data:
U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 12, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 13, /* reset=*/ 43);

This produces a correctish display - but sort of "wraps" by 3 or 4 pixels.  Also the graphics are VERY slow.  I am going to do more digging.  This seems to me like there may be an issue with the default hardware spi pins for this board.

Fishbone69, 2023-08-29:
*************************
Follow-up
*************************
Tried a different esp32-s3 mini microcontroller (same brand) - screen still garbled.
Tried a different SH1106 display - screen still garbled.
The fact that the software SPI constructor for SSD1306 worked is important.

olikraus, 2023-08-29:
U8g2 will use the default pins from the Arduino Environment: Same pins as SPI object would use.

Fishbone69, 2023-08-29:
Another difference between the esp32-S3 and C3 is the cpu clock speed (240MHz on the S3 vs 160 MHz on C3)...could that be relevant?

****Note**** I set the cpu speed at 160 MHz on the S3 and still didn't work....very frustrating.  I am at a loss.

Fishbone69, 2023-08-29:
Dumb question:  Do I have to set pinMode for any of the SPI pins or does the constructor do that?   I didn't have to do that before...reaching for straws.

Fishbone69, 2023-08-29:
[image]
The black "band" is from the camera and is not visible.

olikraus, 2023-08-29:
> Dumb question: Do I have to set pinMode for any of the SPI pins or does the constructor do that? I didn't have to do that before...reaching for straws.

No, this is done by u8g2

> The black "band" is from the camera and is not visible.

If the display shows up with some content, then at least some communication has happend. I assume there is a wiring issue. Especially heck the A0 (=CD) wire.

In general: Avoid using such protoboards. They are very unreliable.

Fishbone69, 2023-08-29:
I understand but here it is with this constructor: U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 12, / data=/ 11, / cs=/ 10, / dc=/ 13, / reset=*/ 43);
[image]

It is not a SSD1306, but I have a hard time thinking this is wiring if I change the constructor and get these results.

olikraus, 2023-08-30:
> At any rate, I changed this pin to any other in the constructor and it seems to work. I would like to know if in general MISO != DC or if it is just this DEV board? Maybe someone can shed some light. I recommend leave the status OPEN for comments.

Like written in the comment, u8g2 does not require any information from the display. As a consequence the MISO input wire to the uC is not required. However some uC will occupy both MISO and MOSI pins once the hardware driven SPI is activated.
It is like this: You want hardware accelerated SPI with MOSI only, but you also get MISO pin occupied. As a consequence the DC pin can not be used on the MISO pin, they would distrub each uther (DC = Data Command output line has nothing todo with MISO SPI line)
I can not speak for the ESP 32, but at least this is also true for AVR systems (Arduino R3), where we can not use pin 12 (which is the MISO pin on AVR systems).
Note: With software emulated SPI, this problem is bypassed and the pin can be used.

> It is not a SSD1306

Indeed, it seems to be a SH1106

Fishbone69, 2023-08-30:
There are a couple of other pins used on the ESP32 in SPI...particularly for use of SPI with Flash memory like FSPIWP (write protect) and FSPIHD.  Those do not seem to cause a conflict  when used as DC output for the display...but I'm thinking if I have extra pins....maybe I avoid those as well?  

Ollie - sorry to cause a fuss.  Your explanation makes perfect sense and has been a real learning experience for me.

Thank you for all that you do!

Fish