olikraus/u8g2 #2255: Library U8G2 on Arduino Portenta Machine Control

jbgraignic, 2023-08-31:
Hello, 

I'm trying to use the library U8G2 to communicate between a Midas display (MD256128A6W-FPTLW) (his controller ST75256) and an Arduino Portenta Machine Control with the protocol I2C. At the moment, I do not find a constructor for my display. Maybe it's useless because the library is not able to work with the ARM Core of the Portenta Machine Control.
This is the wiring diagram between the two components.
Thank you or your help.

[image]

olikraus, 2023-08-31:
Which u8g2 constructor did you use? How does you code look like? Which arguments did you use for the constructor?

jbgraignic, 2023-09-01:
I not find any constructor who seems to work with the portenta. I looking for one of them with ST75256 and I2C. At the moment, I am just trying to make it work an Hello World example (full buffer).
Thank you for your help.

olikraus, 2023-09-01:
All u8g2 constructors work with the Arduino IDE. If you use Arduino IDE also with Portenta, then I don't see a reason why the u8g2 constructors shoudn't work. 

You need to select a u8g2 constructor for your display, the a proper wiring and then configure the parameters of the constructor to match your wiring.

This is all I can say. For further help, past your none-working code here.

olikraus, 2023-09-05:
> The only constructor who uses I2C and ST75256 is...

There are many more, please check here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx256128

In fact I suggest to start with the following constructor `U8G2_ST75256_JLX256128_F_SW_I2C(U8G2_R0, clock, data, reset)`.
Especially you should provide the pin numbers of your Arduino board, where the clock and data line of the I2C bus is connected. You could later switch to HW I2C if required.

Other checks:
- Is the display configured as I2C?
- What is the I2C base address? Did you multiply the address with 2 for u8g2?
- Are pull up resistors added?
- Are the data and clock signal wired correctly? What the pin numbers?