olikraus/u8g2 #2256: Memory problem uses 900+bytes

AexaDe, 2023-08-31:
Im facing a memory problem using uglib2 it uses to much memory (120%) of the available 2048bytes. when i remove the library from my sketch it occupies only 20%. Am i doing anything wrong? is there a way to make it more lightwheight i just need to print some text on a 128x64 oled screen i2c. I tried using adafruit's library but it has serious timing issues and crashes my Arduino.
Thank you in advance.

olikraus, 2023-08-31:
U8g2 has a low memory mode, called "page mode".
Which u8g2 constructor did you use? I suggest to use the  `_1_` constructor in your case. However note, that the graphics output is little bit different, see the page mode examples.

AexaDe, 2023-08-31:
Thank you so much for your reply, i really appreciate your work.
This was the constructor i was using U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
Note that i dont need to do anything fancy like graphics and custom sprites i just wanted plain text. Thank you gain.

olikraus, 2023-08-31:
As I said, try to use the "1" constructor: U8G2_SSD1306_128X64_NONAME_1_HW_I2C 

But you need to change your software then, because the send buffer command will not work as expected.

> i just wanted plain text.

Then maybe the u8x8 constructor then:

U8X8_SSD1306_128X64_NONAME_HW_I2C

But the commands are again different.

AexaDe, 2023-08-31:
thank you, issue solved!

AexaDe, 2023-08-31:
but i have one last question why did you call it page mode instead of low memory mode?

olikraus, 2023-09-01:
This memory area is called a page is almost all display controller datasheets.

closed 2023-09-01 by AexaDe