olikraus/u8g2 #2259: [HELP] Custom fonts

Dario-Ciceri, 2023-09-01:
hi, how to create a custom font from something like this https://www.fontrepo.com/font/21332/seven-segment?

I can't understand how fony works and how to convert bdf (parameters, command line etc)
For example: i would like to have a font with 0123456789+-. size 15x31 what should I do?

olikraus, 2023-09-01:
Did you check the FAQ?

https://github.com/olikraus/u8g2/blob/605f6c5c8e369cd52b87dba61dd47479c9a87c61/doc/faq.txt#L255-L294

Let me know if you have still question.

olikraus, 2023-09-01:
Some 7-segment fonts are listed here: https://github.com/olikraus/u8g2/wiki/fntgrpu8g

Dario-Ciceri, 2023-09-01:
> Did you check the FAQ?
> 
> https://github.com/olikraus/u8g2/blob/605f6c5c8e369cd52b87dba61dd47479c9a87c61/doc/faq.txt#L255-L294
> 
> Let me know if you have still question.

Yes, also read some old issues! It was me not understanding how to use Fony.... now it perfectly works!
Great library!
Thanks.

closed 2023-09-01 by Dario-Ciceri