olikraus/u8g2 #2260: UC1628 parallel driver (128x64)

olikraus, 2023-09-02:
### Discussed in https://github.com/olikraus/u8g2/discussions/2011

This is created from the related discussion: Implement support for UC1628 displays.
As far as I understood the problem, the request is to support two different displays: 128x64 and 256x128

The 256x128 display as the following pinout at the flex cable:
| Pin | Label | Description |
|---|---|---|
| 1 | SDA Out | |
| 2 | SDA In | |
| 3 | SCK | |
| 4 | RST | |
| 5 | CS0 | |
| 6 | CD | |
| 7 | VSS | 0V, GND |
| 8 | VDD | | 
| 9 | VLCD In | Connect to VLCD Out, 1uF to GND |
| 10 | VLCD Out | Connect to VLCD In, 1uF to GND |
| 11 | VB0- | 1uF to VB0+ |
| 12 | VB1- | 1uF to VB1+ |
| 13 | VB1+ | 1uF to VB1- |
| 14 | VB0+ | 1uF to VB0- |

olikraus, 2023-09-02:
Init sequence for the 256x128 display, second arg is the CD level.
```
   RST=0;
   Delay(500);
   RST=1;
   Delay(5000);
   WriteData(0X10,0);     //temp vs VLCD
   WriteData(0X12,0);     //Temp  VS frame rate
   WriteData(0X14,0);     //Temp sensor  control
   WriteData(0X16,0);
   WriteData(0X00,1);
   WriteData(0X0d,1);
   WriteData(0X40,0);    //SET SCROLL LINE
   WriteData(0,1);		 //SL= COM 0
   WriteData(0X20,0);    //11.SET TEMPERATURE COMPENSATION 0X24---0x20
   WriteData(0X2D,0);    // INTERNAL Vlcd
   WriteData(0XEB,0);    //bias=1/12 (28)
   WriteData(0X81,0);    //SET PM
   WriteData(0X00,1);
   WriteData(0X56,1);   //14.5v
   WriteData(0Xc2,0);    //23.SET LCD MAPPING CONTROL
   WriteData(0Xb8,0);    //MTP disable
   WriteData(0X00,1);    //MTP disable
   WriteData(0Xf1,0);    //31.SET COM END
   WriteData(0X7F,1);    //127
   WriteData(0Xf2,0);    //32.SET PARTIAL DISPLAY START()
   WriteData(0X00,1);      //SEG0
   WriteData(0Xf3,0);    //33.SET PARTIAL DISPLAY END
   WriteData(0X7F,1);    //S127	 
   WriteData(0X85,0);    //18.SET PARTIAL DISPLAY CONTROL	 
   Delay(2000);
   WriteData(0XC9,0);
   WriteData(0XAD,1);
```

olikraus, 2023-09-04:
[image]

olikraus, 2023-09-04:
@https://github.com/schmidtas

I have created a beta release 2.35.5 with support for the 256x128 display:
```
U8G2_UC1628_256X128_1_4W_SW_SPI
U8G2_UC1628_256X128_1_4W_HW_SPI
U8G2_UC1628_256X128_F_4W_SW_SPI
U8G2_UC1628_256X128_F_4W_HW_SPI
...
```

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2023-09-09:
@fuyuqingfeng @ullahsh @bhavyasrig @shlyakpavel 

I have created a 128x64 variant of the UC1628 based display based on the code for a 256x128 display with the same controller.
The implementation may not work yet, but it could be a good starting point. However I need your feedback, whether and how it works as of now. 

A subset of the constructors are also added to all examples:
``` C++
U8G2_UC1628_128X64_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7,  /*enable=*/ 18, /*cs=*/ 3, /*dc=*/ 16, /*reset=*/ 17); // UC1628 pin names: enable: /WR, dc: CD
U8G2_UC1628_128X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_UC1628_128X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_UC1628_256X128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_UC1628_256X128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```
Of course, there is also a 8080 constructor for the 256x128 display, it is just not listed in the examples.

The new code is available in beta release 2.35.6. Maybe someone can test this 128x64 display and give some feedback.
Testing goal: We should be able to see something. Any kind of black pixel (garbage, font, anything) would be great.
You can test both the 256x128 or the 128x64 driver to see something.
As a test, use the ContrastTest and the FlipMode examples from the Arduino IDE.
Please send a picture of at least some black pixel are visible.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2023-10-02:
Hi All
I have added support for the 256x32 display from @shlyakpavel . Thanks a lot for sending me the display.

Let me know if there is anything else to do.

I will close this issue as done...

shlyakpavel, 2024-01-16:
@olikraus thanks a lot for supporting the display module!
It works for me on both Linux and esp32
However, the linux port sets CS pin to high every time I run the app and I could not track down the reason/origin of this behavior. The display works fine as long as CS pin on the display module is connected to GND instead of GPIO CS pin.
Any ideas?

olikraus, 2024-01-19:
Thanks for your feedback and the display.

For the linux port I can not say much... it was (most probably) not written by me. Maybe the linux port is wrong and needs some update.

closed 2024-08-24 by olikraus