olikraus/u8g2 #2262: Latest version not available in Arduino Library manager?

pe1rmg, 2023-09-07:
Hello,

I am trying to install the latest version of the library, since I have a JLX 320x160 display which has been added very recently, but it is not available yet in the Arduino library manager.
What is normally the delay between the release of a new version, and the availability in the Arduino library manager? 

Thanks for the great library!

olikraus, 2023-09-07:
I assume you refer to this issue: https://github.com/olikraus/u8g2/issues/2186

It is assigned to current release. 

You could try the latest beta..

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

pe1rmg, 2023-09-08:
Thanks for the information. I was able to get the library installed. Unfortunately it is not including the constructor for the display I have. It uses the same pinout as refered to in the message https://github.com/olikraus/u8g2/issues/2186#issuecomment-1615322143

[image]
[image]

Would it be possible to get the constructor included in the next (beta) release?

Thanks again!

olikraus, 2023-09-08:
Did you check this page: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7586s-jlx320160 ?
All the constructors are already added.
Not sure what else I could do here.

pe1rmg, 2023-09-08:
OK, my bad.  I did install the last Beta version, but then the Arduino IDE decided to "update" my library to 2.34.xx .... Now I have reinstalled 2.35.5, and it does compile now...

My apologies for wasting your time!

olikraus, 2023-09-08:
No problem