olikraus/u8g2 #2263: STM32G030 with u8g2 library problem

sunnyguhz, 2023-09-14:
Hi I have an ST7567 LCD panel that has a 4-pin SPI connection, and I use U8g2 library.

The panel works fine with Arduino(ESP8266 or STM32F103C8T6) under Hardware and Software SPI mode.

It displays okay in Software SPI mode with STM32G030C8T6.
```
#include <U8g2lib.h>
#include <SPI.h>

//U8G2_ST7567_OS12864_1_4W_HW_SPI u8g2(U8G2_R2,  /* cs=*/ PA4, /* dc=*/ PA6, /* reset=*/ PB1);  
U8G2_ST7567_OS12864_1_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/PA1, /* data=*/ PA2, /* cs=*/ PA4, /* dc=*/ PA6, /* reset=*/ PB1); 

void setup(void) {
  u8g2.begin(); 
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}
```

But it it just won't show anything in hardware SPI mode.
```
#include <U8g2lib.h>
#include <SPI.h>

U8G2_ST7567_OS12864_1_4W_HW_SPI u8g2(U8G2_R2,  /* cs=*/ PA4, /* dc=*/ PA6, /* reset=*/ PB1);  
//U8G2_ST7567_OS12864_1_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/PA1, /* data=*/ PA2, /* cs=*/ PA4, /* dc=*/ PA6, /* reset=*/ PB1); 

void setup(void) {
  u8g2.begin(); 
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}
```

OS: [ Windows10]
Arduino IDE version: [2.2.0]
STM32 core version: [2.6.0]
U8g2 version: [2.34.22]
Tools menu settings if not the default: [Debug:None, Optimize:Smallest(-Os) with LTO, USB:None, U(S)ART:Disable]
Upload method: [SWD]

olikraus, 2023-09-14:
Do you use STMduino? 
But in general, U8g2 HW SPI will just use the SPI object from your base software. So you might need to check the base software, which might be STMduino if you use the same.

sunnyguhz, 2023-09-24:
> Do you use STMduino? But in general, U8g2 HW SPI will just use the SPI object from your base software. So you might need to check the base software, which might be STMduino if you use the same.

I used [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) , and I don't know which one has the problem between U8g2 or Arduino_Core_STM32.

olikraus, 2023-09-24:
I assume there is a wiring problem then...

sunnyguhz, 2023-09-25:
> I assume there is a wiring problem then...

Hi, I have found the cause of this problem.
My chinese net friend tell me that MISO cannot be occupied even if MISO not need for LCD panel. And it all so can be change to another pins like PA11 or PB4.