olikraus/u8g2 #2267: U8G2 and timer interrupts

robmaxtech, 2023-09-19:
I'm using U8G2 on an AVR128DB48 with an 128x32 OLED; works excellent! But one question is open for me: I'm using a timer interrupt for audio playback which works fine and without interfering any UI functionality; but updating the display blocks the timer interrupt obviously. Does the U8G2 lib disables interrupts or something similar? I couldn't find anything in the code, but I'd like to understand why the timer interrupt routine is affected. Thanks a lot!

robmaxtech, 2023-09-20:
Ok, thanks; good to know! So I know I don't have to suspect U8G2 any more ;-)  It's such a great library!

closed 2023-09-20 by robmaxtech

olikraus, 2023-09-20:
Of course U8g2 will use SPI or Wire interface as implemented by the Arduino Environment. Those low level communication procedures may use interrupts.