olikraus/u8g2 #2268: NodeMcu-32S hw I2C no way to detect ssd1309

kunpoolsion, 2023-09-20:
Hi, 
First of all: thanks for all the work with the libraries.
I have read everything I was able to find to solve this, without success.
Hardware:
NodeMcu-32S - ESP-WROOM-32 (SCL 21, SDA 22) (tried with different boards, same and different models, but all ESP32).
Oled from Diymore SSD1309 128x64 I2C.

I tried lots of constructors ssd1306 and 1309, trying to specify the pins as 22 and 21 in constructor and/or in Wire, tried adding the addres (0x78 as writen in pcb, 0x3c, 0x3d, 0x7a...), change frequency, add resistors (tried 2.xK with 3.3v, 4.7K with 3.3v and 5v, and 10K with 5v).... everything I think.
Scanners doesn´t detect the oled neither.
Tried example sketch, u8g2 and u8x8.

With 1306 and 1309 SW constructors works properly without and with all resistors combination, but slow.

Any advice to try something alse before give up?
Thanks

olikraus, 2023-09-20:
The SW I2C constructors do work, correct? What is the full constructor line for the SW_I2C constructor?

> NodeMcu-32S - ESP-WROOM-32 (SCL 21, SDA 22)

According to this picture the SCL and SDA lines are swapped:

https://europe1.discourse-cdn.com/arduino/original/4X/9/9/e/99ecaa793c38834e88f0686ee7445ffcc773653f.jpeg

kunpoolsion, 2023-09-20:
Hi Olikraus,
Thnaks for answer.
My bad writting the issue, I was using the correct pins as you wrote, 21 SDA, 22 SCL.

About SW I2C constructor it works with a lot of them, but as example:

U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8( SCL, SDA, U8X8_PIN_NONE );
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8( 22, 21, U8X8_PIN_NONE );
U8X8_SSD1309_128X64_NONAME0_SW_I2C u8x8( SCL, SDA, U8X8_PIN_NONE );
U8X8_SSD1309_128X64_NONAME2_SW_I2C u8x8( SCL, SDA, U8X8_PIN_NONE );
U8X8_SSD1309_128X64_NONAME0_SW_I2C u8x8( 22, 21, U8X8_PIN_NONE ); 
U8X8_SSD1309_128X64_NONAME2_SW_I2C u8x8( 22, 21, U8X8_PIN_NONE );

These constructors works with or without Wire.h enabled.

olikraus, 2023-09-21:
Can you try the pin remapping feature of the constructor for HW I2C?
..._HW_I2C u8x8(U8X8_PIN_NONE, 22, 21);

Oliver

kunpoolsion, 2023-09-21:
Hi Oliver,
Thanks for reply,

I tried your advice, with and without wire.h enabled, no luck.
I tried the Hello world full_buffer example too, with U8G2 constructor (U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ 16, /* data=*/ 17);   // ESP32 Thing, HW I2C with pin remapping).

Still no luck in any case :(

olikraus, 2023-09-21:
Then... I don't know. I think it is now required to trace the I2C signals with a scope. Are the signals present at the pins of the ESP-WROOM-32? Are the signals present on the display?

kunpoolsion, 2023-09-21:
As long as the SW constructor can be used on same pins, I thought all pins are working properly, I mean soldering level.
Sadly I have no scope here.

Time to buy other displays or switch to SPI?

olikraus, 2023-09-21:
I personally think that there is an issue with the Wire lib implementation for ESP32... BTW, I assue that you use the Arduino IDE, is this correct?

kunpoolsion, 2023-09-21:
is correct

olikraus, 2023-09-21:
ok... a scope would be fine, but as an alternative test you could try to create a simple example, which does some I2C communication with Wire lib. But you might require a I2C client for this...

kunpoolsion, 2023-09-21:
With scope you mean an osciloscope or multimeter? I have multimeter, of course.
Can I use other esp32 as client?

olikraus, 2023-09-21:
Yes osciloscope. You could also use the display as a client.

kunpoolsion, 2023-09-21:
how can I do the test?

kunpoolsion, 2023-09-21:
Definitely must be the display.
I took out a DS3231 module from other project and the i2c scanner can fund it.
You can close the issue, already ordered another display.
Thanks a lot

olikraus, 2023-09-21:
Oh, i just remember your initial post: Scanners doesn´t detect the oled neither.
There is an odd topic with those displays. You need to connect the two lines SDAin and SDAout on the PCB. I am surprised, that those broken displays are still sold.

The problem is for example described here: https://www.eevblog.com/forum/beginners/0-96-oled-display-i2c-config/msg4169437/?PHPSESSID=npi9apg3qo8f8se5g53hjfot4d#msg4169437

The SW_I2C constructor still works with u8g2, because u8g2 ignores the missing ACK signal with software emulated I2C. However HW_I2C requires a valid ACK signal.

kunpoolsion, 2023-09-22:
ummmmm....
I´m good on soldering, I will try making a jumper between D1 and D2 and check again.
There is a way to skip the ack back in the wire library? (maybe thats too deep to go)
Thanks oliver for the help. I´ll be back with a report.

kunpoolsion, 2023-09-22:
Nahhhhh....
tested, D1+D2 not recognized by scanner nor HW constructor
removed pull down resistor from D1+D2, neither
Giving up, will wait for new displays

kunpoolsion, 2023-09-22:
Ok, finally fixed, before the SDA pin there was a diode, if you short the diode, _voilá_
soooo... finally fixed.
Thanks Oliver

olikraus, 2023-09-23:
For u8g2 / u8x8 the i2c address has to be multiplied with 2, see the manual...