olikraus/u8g2 #2274: DC pin not being asserted - Atmega1608 port.

andreikovari, 2023-09-30:
Setup: SSD1306 display connected through SPI (4 wire mode) to Atmge1608. Programming though UPDI.

I did a SW SPI test and works, display works as expected. 
Trying to use a HW SPI approach, data is being sent correctly, RES and CS are being asserted as expected, but DC is being kept in a low state.

`uint8_t u8x8_byte_avr_4wire_hw_spi (u8x8_t * u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	uint8_t *data;

	switch (msg) {
		
		case U8X8_MSG_BYTE_INIT:
		u8x8->bus_clock = u8x8->display_info->sck_clock_hz;		
		u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);		
		
		case U8X8_MSG_BYTE_SET_DC:
		u8x8_gpio_SetDC(u8x8, arg_int);
		break;
		
		case U8X8_MSG_BYTE_START_TRANSFER:
		u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
		u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
		break;
		
		case U8X8_MSG_BYTE_SEND:
    data = (uint8_t *)arg_ptr;
      while( arg_int > 0 ) {
        SPI0.DATA = (uint8_t)*data;
        while (!(SPI0.INTFLAGS & SPI_IF_bm));
        data++;
        arg_int--;
      }  
      break;	
		
		case U8X8_MSG_BYTE_END_TRANSFER:
		u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
		u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
		break;
		default:
		return 0;
	}
	
	return 1;
}

uint8_t u8x8_avr_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t cycles;

  switch (msg)
  {
  case U8X8_MSG_DELAY_NANO: // delay arg_int * 1 nano second
    // At 20Mhz, each cycle is 50ns, the call itself is slower.
    break;
  case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
                               // Approximate best case values...
#define CALL_CYCLES 26UL
#define CALC_CYCLES 4UL
#define RETURN_CYCLES 4UL
#define CYCLES_PER_LOOP 4UL

    cycles = (100UL * arg_int) / (P_CPU_NS * CYCLES_PER_LOOP);

    if (cycles > CALL_CYCLES + RETURN_CYCLES + CALC_CYCLES)
      break;

    __asm__ __volatile__(
        "1: sbiw %0,1"
        "\n\t" // 2 cycles
        "brne 1b"
        : "=w"(cycles)
        : "0"(cycles) // 2 cycles
    );
    break;
  case U8X8_MSG_DELAY_10MICRO: // delay arg_int * 10 micro seconds
    for (int i = 0; i < arg_int; i++)
      _delay_us(10);
    break;
  case U8X8_MSG_DELAY_MILLI: // delay arg_int * 1 milli second
    for (int i = 0; i < arg_int; i++)
      _delay_ms(1);
    break;
  default:
    return 0;
  }
  return 1;
}

uint8_t u8x8_avr_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  // Re-use library for delays

  switch (msg)
  {
  case U8X8_MSG_GPIO_AND_DELAY_INIT: // called once during init phase of u8g2/u8x8
      DISPLAY_PORT.DIR |= DISPLAY_CLK_PIN;   
      DISPLAY_PORT.DIR |= DISPLAY_DATA_PIN;   
      DISPLAY_PORT.DIR |= DISPLAY_CS_PIN;   
      DISPLAY_PORT.DIR |= DISPLAY_DC_PIN;   
      DISPLAY_PORT.DIR |= DISPLAY_RESET_PIN;  
    break;                      // can be used to setup pins
  case U8X8_MSG_GPIO_CS: // CS (chip select) pin: Output level in arg_int
    if (arg_int)
      DISPLAY_PORT.OUT |= DISPLAY_CS_PIN;
    else
      DISPLAY_PORT.OUT &= ~DISPLAY_CS_PIN;
    break;
  case U8X8_MSG_GPIO_DC: // DC (data/cmd, A0, register select) pin: Output level in arg_int
    if (arg_int)
      DISPLAY_PORT.OUT |= DISPLAY_DC_PIN;
    else
      DISPLAY_PORT.OUT &= ~DISPLAY_DC_PIN;
    break;
  case U8X8_MSG_GPIO_RESET: // Reset pin: Output level in arg_int
    if (arg_int)
      DISPLAY_PORT.OUT |= DISPLAY_RESET_PIN;
    else
      DISPLAY_PORT.OUT &= ~DISPLAY_RESET_PIN;
    break;
  default:
    if (u8x8_avr_delay(u8x8, msg, arg_int, arg_ptr)) // check for any delay msgs
      return 1;
    u8x8_SetGPIOResult(u8x8, 1); // default return value
    break;
  }
  return 1;
}

void u8g2_setup(u8g2_t *u8g2)
{
    SPI0.CTRLA = 0b00110101;
    SPI0.CTRLB = 0b00000100;
    _delay_ms(100);
    u8g2_Setup_ssd1306_128x32_univision_f(u8g2, U8G2_R0, u8x8_byte_avr_4wire_hw_spi, u8x8_avr_gpio_and_delay);
}`

There were no changes on the DC setup or control side from the SW variant. Is there another flag that would set the SPI mode in 3wire or 4wire SPI mode?

Any feedback or insight is appreciated!

olikraus, 2023-09-30:
Code looks ok to me. Be arware, that you must not use MISO or any other pins connected to the HW SPI subsystem when HW SPI is active. I think we had such a case recently here in this forum.

andreikovari, 2023-09-30:
Actually MISO pin is used as DC. Maybe SPI is overwriting the DC as being an output and then the library cannot control it anymore?

andreikovari, 2023-09-30:
[image]
I think this might be it. 
Thanks for pointing me in the right direction! 
Issue can be closed. 
Resolution: DC pin is used instead of MOSI. Using HW SPI overwrite any setting to MOSI, thus the pin cannot be used as an output anymore.