olikraus/u8g2 #2276: u8g2 porting to STM32F103 /SSD1306_128X64_SW_I2C

keyone808, 2023-10-04:
I've been having some trouble lately.  I must try to port to stm32f103 standard library, when I want to use u8g2.  But I find that even if I reference the u8g2 library, I still can't use the functions in his library.
[image]

olikraus, 2023-10-04:
Hmmm not sure how we can help here. The symbols are also not part of u8g2 lib.

keyone808, 2023-10-04:
> Hmmm not sure how we can help here. The symbols are also not part of u8g2 lib.

sure? OK, I thought that those function are part of u8g2 lib. Then I'll find out what the problem is. Thank you !!!

miaowu123, 2023-10-05:
[image]
你使用的问题

keyone808, 2023-10-06:
卧槽谢谢！！！！

keyone808, 2023-10-06:
> [image] 你使用的问题

哥，问些问题

keyone808, 2023-10-06:
> [image] 你使用的问题

哥问一些问题，1.我发现U8G2的所有库里面的函数都有u8g2这个结构体，我是需要单独定义一个才可以用它里面的函数嘛？2.我该怎么知道这个库该怎么用，怎么驱动SSD1306这个屏幕（就一些初始化函数我该通过怎么样的途径找到教程！！球球球帅哥告知）---好人一生平安

njuFerret, 2023-10-10:
> > [image] 你使用的问题
> 
> 哥问一些问题，1.我发现U8G2的所有库里面的函数都有u8g2这个结构体，我是需要单独定义一个才可以用它里面的函数嘛？2.我该怎么知道这个库该怎么用，怎么驱动SSD1306这个屏幕（就一些初始化函数我该通过怎么样的途径找到教程！！球球球帅哥告知）---好人一生平安

库里面一堆例程，慢慢看