olikraus/u8g2 #2278: 3.6" TFT, ST7793, 240x400

red-scorp, 2023-10-14:
I've recently bought an Arduino shield for tests with following specs:

3.6 Inch Resolution 240x400 Driver ST7793 Arduino UNO and Mega2560 Resistive Touch LCM Module

Product Specification:
•	Supports development boards such as Arduino UNO and Mega2560 for plug-in use without wiring
•	400X240 resolution, clear display, support for touch function
•	Support 16-bit RGB 65K color display, display rich colors
•	8-bit parallel bus, faster than serial SPI refresh
•	On-board 5V/3.3V level shifting IC, compatible with 5V/3.3V operating voltage
•	Easy to expand the experiment with SD card slot
•	Provides an Arduino library with a rich sample program
•	Military-grade process standards, long-term stable work
•	Provide underlying driver technical support

| Name | Parameter |
| --- | ---|
| Display Color | RGB 65K color |
| SKU | MAR3601(with resistive touch screen) |
|Screen Size | 3.6(inch) |
| Type | TFT |
| Driver IC | ST7793 |
| Resolution | 400*240 (Pixel) |
| Module Interface | 8-bit parallel interface |
| Active Area | 90.2*52.5(mm) |
| Module PCB Size | 77.40*46.44(mm) |
| Operating Temperature | -20ºC~70ºC |
| Storage Temperature | -40ºC~70ºC |
| Operating Voltage | 5V/3.3V |
| Power Consumption | TBD |
| Product Weight(Package containing) | TBD |

The only reasonable information of the shield I was able to find here:
https://ronboelectronics.en.made-in-china.com/product/rOktflPcYTVe/China-3-6-Inch-Resolution-240X400-Driver-St7793-Arduino-Uno-and-Mega2560-Resistive-Touch-LCM-Module.html

Few questions then:
1. How can I hook up this screen to UNO/MEGA with u8g2?
2. If unsupported for now, is there a porting guide available?

Thanks for you answers in advance!

olikraus, 2023-10-14:
U8g2 only supports monochrome Displays

red-scorp, 2023-10-14:
What other arduino libraries can support this display?

olikraus, 2023-10-22:
good question, maybe google helps?

closed 2023-10-27 by red-scorp

DzoniSkupnjak, 2024-01-18:
Hi,
Can I use the same display as mentioned with UTFT and URTouch libraries?

Thanks!

Gribouille28, 2024-03-01:
MCUFRIEND_kbv.h is the only library to my knowledge hat currently supports the combination Arduino Uno and ST7793 controller.

Bye

danielw1990, 2024-04-04:
Blogpost about this:
https://thesolaruniverse.wordpress.com/2023/11/03/uno-shield-3-6-inch-240400-pixel-tft-display-with-st7793-controller-2/
The blog also mentions MCUFRIEND_kbv.h as the library used. But also explicitly states not having tested the touch features.

it also contains an Arduino Sketch I have not yet tested. I just found this issue during research before purchase.

Yup, that's all I got. Good luck.

DzoniSkupnjak, 2024-04-04:
Thank you very much!!

Avtollic, 2024-04-07:
всім доброго часу! підкажіть будь ласка як підєднати ST7793 до ESP8266 nodemcu? завчасно вдячний за відповідь!

Mercorreia, 2024-10-02:
ESTA TELA SO FUNCIONA COM ESTA BIBLIOTECA:

 MCUFRIEND_kbv.h 

SEGUE CODIGO EXEMPLO:

#include <MCUFRIEND_kbv.h>  // Biblioteca da tela TFT

MCUFRIEND_kbv tft;

#define BLACK 0x0000
#define WHITE 0xFFFF
#define BLUE  0x001F

const int potPin = A0;  // Pino onde o potenciômetro está conectado
int potValue = 0;  // Variável para armazenar o valor lido do potenciômetro

void setup() {
  Serial.begin(9600);
  
  uint16_t ID = tft.readID();  // Lê o ID do controlador da tela
  tft.begin(ID);  // Inicializa o display com o ID correto
  tft.setRotation(1);  // Define a rotação da tela (1 para paisagem)
  
  tft.fillScreen(BLUE);  // Preenche a tela com a cor azul
  
  // Configurações iniciais de texto
  tft.setTextSize(3);  // Tamanho do texto
  tft.setTextColor(WHITE);  // Cor do texto
  tft.setCursor(40, 50);  // Define a posição do texto (x, y)
  tft.print("Te amo, Emily");  // Exibe a mensagem inicial
}

void loop() {
  potValue = analogRead(potPin);  // Lê o valor do potenciômetro
  
  // Atualiza o valor do potenciômetro na tela
  tft.setTextSize(2);  // Tamanho do texto para o valor
  tft.setTextColor(WHITE, BLUE);  // Texto branco com fundo azul (para sobrescrever o texto antigo)
  tft.setCursor(40, 120);  // Posição do valor do potenciômetro
  tft.print("Valor: ");
  tft.print(potValue);  // Exibe o valor do potenciômetro
  
  delay(500);  // Aguarda meio segundo antes de atualizar novamente
}