olikraus/u8g2 #2279: uc1601 spi draw Incorrect，please help!!!

myvcgit, 2023-10-15:
i use 
u8g2_Setup_uc1601_128x64_f(&u8g2, U8G2_MIRROR, u8x8_byte_3wire_hw_spi,  u8g2_gpio_and_delay_stm32);

when use drawStr is Incorrect display。please help !!!

u8g2_SetFont(&u8g2, u8g2_font_t0_11b_tr); 
u8g2_DrawStr(&u8g2, 2, 60, "Hello World!");

[image]

u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr); 							 
u8g2_DrawStr(&u8g2, 2, 60, "Hello World!");	

[image]

when i modify uint8_t u8x8_d_uc1601_128x64_init_seq
static const uint8_t u8x8_d_uc1601_128x64_init_seq[] = {
  
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x0e2),            			/* soft reset */
  U8X8_C(0x02f),            			
	

  //U8X8_C(0x0ea),            			/* LCD Load + Internal Charge Pump (default: 0x2e) */
  U8X8_C(0x081),		                
  U8X8_C(0x0af),		                
  U8X8_C(0x0f1),		                
  U8X8_C(0x0a0),		                
//  U8X8_CA(0x081, 0x0df),		/* set contrast */
//  U8X8_C(0x02e),            			/* LCD Load + Internal Charge Pump (default: 0x2e) */	
//  U8X8_C(0x040),		                /* set display start: See Issue 1501 */
  
  U8X8_C(0x0a6),		                /* normal display */
   
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

u8g2_SetFont(&u8g2, u8g2_font_t0_11b_tr); 
u8g2_DrawStr(&u8g2, 2, 3, "Hello World!");
[image]

u8g2_DrawStr(&u8g2, 2, 8, "Hello World!");
[image]

u8g2_DrawStr(&u8g2, 2, 36, "Hello World!");
[image]

u8g2_DrawStr(&u8g2, 2, 40, "Hello World!");
[image]

u8g2_DrawStr(&u8g2, 2,60 , "Hello World!");
[image]

----------------------------------------------------------
u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr); 							 
u8g2_DrawStr(&u8g2, 2, 40, "Hello World!");	
[image]

u8g2_DrawStr(&u8g2, 2, 60, "Hello World!");	
[image]

olikraus, 2023-10-22:
So you solved the problem by changing the init sequence?

myvcgit, 2023-11-09:
> So you solved the problem by changing the init sequence?

no ，changing sequence can display,but the position  Incorrect.

olikraus, 2023-11-23:
hmmm... not sure how to continue here. From where did you get the display?