olikraus/u8g2 #2286: U8g2 not working but u8X8 working OGM12864-6FLW3/X display

sudheerbussa, 2023-10-26:
I am interfacing the said GLCD module. It has ST7565 driver in parallel data mode. It has no SPI or I2C support as per their datasheet. Tried all variants of constructors of ST7565_6800. But ERC12864_ALT giving some output with U8X8. But nothing is coming using u8g2. My main requirement is graphics. Not just text. If you give me a fix, it would be of great help.

Attaching datasheet, code used and sample output with U8X8. 

//#include <Arduino.h>
#include <U8x8lib.h>

//#ifdef U8X8_HAVE_HW_SPI
//#include <SPI.h>
//#endif
[OGM12864 6FL3.pdf](https://github.com/olikraus/u8g2/files/13175415/OGM12864.6FL3.pdf)

U8X8_ST7565_ERC12864_ALT_6800 u8x8(11,10,9,8,7,6,5,4,12,17,15,2);//working
//EA_DOGM128,LM6063-not working,64128N- inverted by 180 degrees and x, y position interchange, ZOLEN_128X64- mirror image and x,y interchange
//U8X8_ST7565_ERC12864_6800(d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset])// Not working for us

void setup(void)
{
    u8x8.begin();
  u8x8.setPowerSave(0);
  
  
}

void loop(void)
{
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0,0,"Innotronix");
  delay(1000);
  u8x8.drawString(0,1,"Labs");
  delay(1000);
  u8x8.drawString(0,2,"Pvt");
  delay(1000);
  u8x8.drawString(0,3,"Ltd");
  delay(1000);
  u8x8.drawString(0,4,"Hurrey");
  delay(1000);
  

}

sudheerbussa, 2023-10-26:
I have one more GLCD SDGB12864SYG which has driver SBN6400, segment driver SBN0064. I couldn't find suitable constructor for this as well in both libraries. Attaching data sheet
[128x64-cob-graphic-lcd-display SDGB12864-02.pdf](https://github.com/olikraus/u8g2/files/13175468/128x64-cob-graphic-lcd-display.SDGB12864-02.pdf)

olikraus, 2023-10-26:
OGM12864: 
If `U8X8_ST7565_ERC12864_ALT_6800 u8x8(11,10,9,8,7,6,5,4,12,17,15,2);` will work, then `U8G2_ST7565_ERC12864_F_6800(U8G2_R0,11,10,9,8,7,6,5,4,12,17,15,2);` should also work.

SDGB12864SYG: Please check with the KS0108 constructor

sudheerbussa, 2023-10-26:
Thank you for such quick response. I tried U8G2_ST7565_ERC12864_F_6800 u8g2(U8G2_R0,11,10,9,8,7,6,5,4,12,17,15,2); But Not working.
For SDGB12864SYG: tried KS0108 constructor both variations keeping CS0 and CS2 PIN_NONE alternatively. But not working.
U8G2_KS0108_128X64_F u8g2 (U8G2_R0,11,10,9,8,7,6,5,4, /* enable = */ 20, /* dc = */ 19, /* cs0 = */U8X8_PIN_NONE, /* cs1 = */ 12, /* cs2 = */13 , /* reset = */ U8X8_PIN_NONE);
U8G2_KS0108_128X64_F u8g2 (U8G2_R0,11,10,9,8,7,6,5,4, /* enable = */ 20, /* dc = */ 19, /* cs0 = */12, /* cs1 = */ 13, /* cs2 = */U8X8_PIN_NONE , /* reset = */ U8X8_PIN_NONE);

Give me a hint what might have gone wrong. Any suggestions on which file i need to edit from library for both cases.
Appreciate your help

olikraus, 2023-10-26:
SDGB12864SYG: Did you connect V0 and Vout correctly? This is actually not described in the datasheet, so you need to check KS0108 sources for proper wiring.

OGM12864: Hmm I never heard about such an issue. Maybe your test code is wrong. Which Arduino board do you use? What is the full example for the u8g2 test?

sudheerbussa, 2023-10-30:
Thank you for the support. Both the displays are working with U8g2 library. The problem was resolved after connecting RESET pin of GLCD to GPIO instead of keeping it open. Please close the issue

closed 2023-10-30 by sudheerbussa