olikraus/u8g2 #2287: FreeRtos with u8g2

PTT31, 2023-10-28:
#include <Arduino.h>
#include "U8g2lib.h"

void LCDTask(void *pvParameters)
{
    (void)pvParameters;
    U8G2_ST7567_JLX12864_F_4W_HW_SPI u8g2(U8G2_R0, /* clock=EXP2_02_PIN,*/ /* data=EXP2_01_PIN,*/ /* cs=*/21, /* dc=*/4, /* reset=*/0);
    u8g2.begin();
    u8g2.setFont(u8g2_font_ncenB08_tr);

    while (1)
    {
        u8g2.clearBuffer();
        u8g2.drawStr(0, 20, "Hello, FreeRTOS!");
        u8g2.sendBuffer();
        vTaskDelay(pdMS_TO_TICKS(1000)); // Cập nhật màn hình sau mỗi giây
    }
}

void setup()
{
    xTaskCreate(LCDTask, "LCDTask", 10000, NULL, 1, NULL);
    vTaskStartScheduler();
}

void loop()
{
    // Loop rỗng, nhiệm vụ chạy
}

[image]
i use u8g2 esp32 library, it can't work with hardware spi, but works with soft spi

olikraus, 2023-10-28:
As the error indicates, you need to extend wdt timeout

PTT31, 2023-10-29:
I used
esp_task_wdt_init(10000,1);
but it not work

olikraus, 2023-10-29:
I don't know the esp framework, but maybe disable wdt .

PTT31, 2023-10-29:
i used u8g2.nextpage() and i work. thank you.

closed 2023-10-29 by PTT31