olikraus/u8g2 #2289: Changing default I2C HW pins on Raspberry Pi Pico RP2040

MIKHANYA, 2023-11-02:
I'm using Raspberry Pi Pico with [earlephilhower arduino-pico](https://github.com/earlephilhower/arduino-pico) core.
 And screen SSD1309_128X64 on 12,13 Pins. And this initialization line in the code `U8G2_SSD1309_128X64_NONAME0_F_HW_I2C u8g2(U8G2_R0)`
 In Arduino IDE, I decided to simply change the default I2C pins in the file. From default 4,5.
` Users/User/Library/Arduino15/packages/rp2040/hardware/rp2040/3.6.0/variants/rpipico/pins_arduino.h`
```
// Wire
#define PIN_WIRE0_SDA  (12u)//(4u)
#define PIN_WIRE0_SCL  (13u)//(5u)
```

But in PlatformIO I can't do that way, can you tell me how to change them in other way ?

olikraus, 2023-11-02:
There is a command for this, see the reference manual.

larrybud2004, 2024-01-09:
> But in PlatformIO I can't do that way, can you tell me how to change them in other way ?

Did you figure this out?  I can't get this to work with Earl's core. Trying:

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(PIN_WIRE0_SCL, PIN_WIRE0_SDA, U8X8_PIN_NONE);

which translates to pin 5, 4 respectively, but no output on the display. I switch back to the standard core, and change to PIN_WIRE_SCL, PIN_WIRE_SDA, which are the same values (5,4) and the display works.

egnor, 2024-04-13:
What you need to do is configure `Wire` (the Arduino I2C layer) with your pins, and NOT pass them to U8g2. See [the documentation for Earle Philhower's core's I2C interface](https://arduino-pico.readthedocs.io/en/latest/wire.html); you need to do something like this:

```
Wire.setSDA(your_sda_pin);
Wire.setSCL(your_scl_pin);
U8G2_SSD1309_128X64_NONAME0_F_HW_I2C u8g2(U8G2_R0);  // do NOT pass SDA or SCL pins, you MAY pass reset pin
u8g2.begin();
```

This is because [the code in U8g2 to initialize I2C](https://github.com/olikraus/u8g2/blob/b325be786792efece0bdefa7c5df3d86db079271/cppsrc/U8x8lib.cpp#L1340) doesn't know how to supply SDA and SCL pins using that core. (In fact, the ONLY Arduino-based system it knows how to do that on is the ESP32. This should probably be clarified and/or improved.)

It's important NOT to pass SDA and SCL pins to the constructor on the RP2040 (normally it should be harmless if you've otherwise configured `Wire` correctly) because of another unfortunate interaction which I'll write up in a separate bug...

olikraus, 2024-04-14:
I actually like the solution with setSDL and setSCL.
Thanks for providing this.

olikraus, 2024-04-14:
see also: https://github.com/olikraus/u8g2/issues/2425

ChitoKim, 2024-07-02:
You can do the same thing on PlatformIO, just the directories are different.
pins_arduino.h things are inside ~\.platformio\packages\framework-arduinopico\variants.
Otherwise, you can set them on .ino file with setSDA, setSCL things.

Plus, if you use 2nd I2C, you need to define macro "WIRE_INTERFACES_COUNT 2".
In PlatformIO you can define such macros in your platformio.ini

[image]

ChitoKim, 2025-08-02:
https://microcontrollerslab.com/raspberry-pi-pico-pinout-features-programming-peripherals/I
I2C0 – SDA | GP0/GP4/GP8/GP12/GP16/GP20
I2C0 – SCL | GP1/GP5/GP9/GP13/GP17/GP21
I2C1 – SDA | GP2/GP6/GP10/GP14/GP18/GP26
I2C1 – SCL | GP3/GP7/GP11/GP15/GP19/GP27
Maybe because GPIO 14 & 15 are for second I2C? How about trying the pins assigned for I2C0 instead?

olikraus, 2025-12-07:
Not sure whether I can do something here, but thanks for reporting.

egnor, 2025-12-07:
@pellematrose try using Wokwi (https://wokwi.com/) and setting up a little simulation? it can absolutely simulate a Pico connected to a 128x64 OLED running u8g2 for testing, and then you can link to your simulation setup and we can debug together if there's a problem.

(OR it will work in sim but not on your bench, which is also interesting.)

egnor, 2025-12-12:
@pellematrose a win for simulation! the problem is that pins 14 and 15 are on port I2C1 (`Wire1`), not the default port I2C0 (`Wire`). This is precisely what @ChitoKim was saying. When I moved to pins 12 and 13, it worked fine...

[image]

OR to keep using pins 14/15, you use the `U8G2_SSD1306_128X64_NONAME_F_2ND_HW_I2C` class and do the pin setup with `Wire1` instead of `Wire`...

[image]

...hope this helps. (Meanwhile the original issue persists, mostly unrelated.)