olikraus/u8g2 #2290: Request additional constructor for KS0108 ERM19264 display

laserranger3000, 2023-11-04:
Hello,

I'd like to request an additional constructor for this display:
https://www.buydisplay.com/download/manual/ERM19264-3_Series_Datasheet.pdf

The chip select works differently, using only 2 pins instead of 3.
The rest should be identical to other products as far as I can see.

olikraus, 2023-11-23:
hmm... i will add this to the current milestone, but i am not sure when I find time to work on this....

mpiernick, 2024-03-18:
Hi,
I just wrote something like this for the U8g2 Arduino.
I don't know if this is the right way, it could probably be done better.
This screen has CSA and CSB pins, there is no reset pin.
```c
diff --git a/src/U8g2lib.h b/src/U8g2lib.h
index 0cf892f5..efbbcc99 100644
--- a/src/U8g2lib.h
+++ b/src/U8g2lib.h
@@ -22902,6 +22902,14 @@ class U8G2_KS0108_ERM19264_F : public U8G2 {
     u8x8_SetPin_KS0108(getU8x8(), d0, d1, d2, d3, d4, d5, d6, d7, enable, dc, cs0, cs1, cs2, reset);
   }
 };
+//mpier...
+class U8G2_KS0108_CSAB19264_1 : public U8G2 {
+  public: U8G2_KS0108_CSAB19264_1(const u8g2_cb_t *rotation, uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t dc, uint8_t csa, uint8_t csb) : U8G2() {
+    u8g2_Setup_ks0108_csab19264_1(&u8g2, rotation, u8x8_byte_arduino_ks0108, u8x8_gpio_and_delay_arduino);
+    u8x8_SetPin_KS0108(getU8x8(), d0, d1, d2, d3, d4, d5, d6, d7, enable, dc, csa, csb, U8X8_PIN_NONE, U8X8_PIN_NONE);
+  }
+};
+//...mpier
 class U8G2_T7932_150X32_1 : public U8G2 {
   public: U8G2_T7932_150X32_1(const u8g2_cb_t *rotation, uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t dc, uint8_t cs0, uint8_t cs1, uint8_t cs2, uint8_t reset = U8X8_PIN_NONE) : U8G2() {
     u8g2_Setup_t7932_150x32_1(&u8g2, rotation, u8x8_byte_arduino_ks0108, u8x8_gpio_and_delay_arduino);
diff --git a/src/U8x8lib.h b/src/U8x8lib.h
index d15fd618..f61b2191 100644
--- a/src/U8x8lib.h
+++ b/src/U8x8lib.h
@@ -7905,6 +7905,14 @@ class U8X8_KS0108_ERM19264 : public U8X8 {
     u8x8_SetPin_KS0108(getU8x8(), d0, d1, d2, d3, d4, d5, d6, d7, enable, dc, cs0, cs1, cs2, reset);
   }
 };
+//mpier...
+class U8X8_KS0108_CSAB19264 : public U8X8 {
+  public: U8X8_KS0108_CSAB19264(uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t dc, uint8_t csa, uint8_t csb) : U8X8() {
+    u8x8_Setup(getU8x8(), u8x8_d_ks0108_csab19264, u8x8_cad_001, u8x8_byte_arduino_ks0108, u8x8_gpio_and_delay_arduino);
+    u8x8_SetPin_KS0108(getU8x8(), d0, d1, d2, d3, d4, d5, d6, d7, enable, dc, csa, csb, U8X8_PIN_NONE, U8X8_PIN_NONE);
+  }
+};
+//...mpier
 class U8X8_T7932_150X32 : public U8X8 {
   public: U8X8_T7932_150X32(uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t dc, uint8_t cs0, uint8_t cs1, uint8_t cs2, uint8_t reset = U8X8_PIN_NONE) : U8X8() {
     u8x8_Setup(getU8x8(), u8x8_d_t7932_150x32, u8x8_cad_001, u8x8_byte_arduino_ks0108, u8x8_gpio_and_delay_arduino);
diff --git a/src/clib/u8g2.h b/src/clib/u8g2.h
index e0570888..49900b59 100644
--- a/src/clib/u8g2.h
+++ b/src/clib/u8g2.h
@@ -1469,6 +1469,9 @@ void u8g2_Setup_ks0108_128x64_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_ms
 void u8g2_Setup_ks0108_erm19264_1(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
 void u8g2_Setup_ks0108_erm19264_2(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
 void u8g2_Setup_ks0108_erm19264_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
+//mpier...
+void u8g2_Setup_ks0108_csab19264_1(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
+//...mpier
 void u8g2_Setup_t7932_150x32_1(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
 void u8g2_Setup_t7932_150x32_2(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
 void u8g2_Setup_t7932_150x32_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
diff --git a/src/clib/u8g2_d_setup.c b/src/clib/u8g2_d_setup.c
index c57c7875..14575d54 100644
--- a/src/clib/u8g2_d_setup.c
+++ b/src/clib/u8g2_d_setup.c
@@ -7227,6 +7227,17 @@ void u8g2_Setup_ks0108_erm19264_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_
   buf = u8g2_m_24_8_f(&tile_buf_height);
   u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
 }
+//mpier...
+/* ks0108 1 */
+void u8g2_Setup_ks0108_csab19264_1(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
+{
+  uint8_t tile_buf_height;
[... 115 lines cut ...]
+	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 0, NULL);  // 3-->6, issue 631
+	u8x8_ks0108_out(u8x8, &v, arg_ptr);
+	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_END_TRANSFER, 3, NULL);
+      }
+      if ( v.x < 16 )
+      {
+	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 2, NULL);
+	u8x8_ks0108_out(u8x8, &v, arg_ptr);
+	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_END_TRANSFER, 3, NULL);
+      }
+      if ( v.x < 24 )
+      {
+	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 1, NULL); // 6-->3, // issue 631
+	u8x8_ks0108_out(u8x8, &v, arg_ptr);
+	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_END_TRANSFER, 3, NULL);
+      }
+      break;
+    default:
+      return 0;
+  }
+  return 1;
+}
+//...mpier
+
```