olikraus/u8g2 #2291: SSD1306 black screen when font type changed

MPeter07, 2023-11-06:
Hi,

If I modify the font type to a smaller one (everywhere in the program) (to **u8g2_font_ncenB08_tr**), the OLED screen is black, and the program hangs, (even the serial monitor doesn't get input, that's why I thing the whole program freezes).

I would like to change the font size to smaller one, when counter values (and labels) are displaying.
With the "**u8g2_font_ncenB10_tr**" font the program runs perfectly.

What is wrong in the program, or maybe issue with the Arduino UNO or with the OLED module?

Thanks for the answer in advane,

Best regards,
Peter

-----------------
Below the code:
-----------------
#include <Arduino.h>
#include <Wire.h>
#include "Timer.h"
#include <U8g2lib.h>

const int buttonPin = 2;  // the pin that the pushbutton is attached to

int buttonState = LOW;        // current state of the button
int itimerTS1Prev = 0;
char sBuff[7];
int iCounter1 = 128, iCounter2 = 102, iCounter3 = 98;

long TS1_Time = 3;
int TimerTS1On = 1;  

Timer timerTS1(MILLIS);

U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C display(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);// [page buffer, size = 128 bytes]);

void setup() {

  pinMode(buttonPin, INPUT);

  Serial.begin(9600);
  
  randomSeed(analogRead(11));
  Serial.println("Timer stopped.");
  
  Wire.setClock(10000);

  display.begin();
  display.clearBuffer();

  OLEDText(&u8g2_font_ncenB10_tr, "Press Button!", 0, 25, true);
}

void loop() {

  if(timerTS1.state() == STOPPED && buttonState == HIGH && TimerTS1On == 1) {
    //TS1_Time = random(TServoFrom, TServoTo);

    Serial.println("Timer TS1 start.");
    Serial.println(" ");
    timerTS1.start();
    OLEDText(&u8g2_font_ncenB10_tr, "Tb:", 0, 14, false);
    OLEDText(&u8g2_font_ncenB10_tr, "Dn:", 63, 21, false);
    OLEDText(&u8g2_font_ncenB10_tr, "Ms:", 0, 31, false);
  }  

  if(buttonState == HIGH) {
    if (TimerTS1On == 1) itimerTS1SerialMonitor();

    if (TimerTS1On == 1) itimerTS1Check();
  }

  if (digitalRead(buttonPin) == HIGH) {  

    if(buttonState == LOW) buttonState = HIGH;
    else buttonState = LOW;
    delay(100);
 
    if (buttonState == HIGH) {
        Serial.println("-LED on");
        OLEDText(&u8g2_font_ncenB10_tr, " ", 0, 14, true);
    } else {
        if (TimerTS1On == 1) timerTS1.stop();
       
        Serial.print("TS1 stopped!");
        Serial.println(" -LED off");
        Serial.println("Press Button.");
        OLEDText(&u8g2_font_ncenB10_tr, "Press Button!", 0, 25, true);
   }
   delay(20);
  }
}

int itimerTS1SerialMonitor() {
    int itimerTS1 = timerTS1.read()/1000;
    if(itimerTS1Prev != itimerTS1 && TS1_Time >= itimerTS1) { 
      Serial.print("Timer TS1: ");
      Serial.print(TS1_Time);
      Serial.print("  -> Timer TS1: ");
      Serial.println(itimerTS1);
      itimerTS1Prev = itimerTS1;
      
      sprintf(sBuff,"%03d", iCounter1--);
      OLEDText(&u8g2_font_ncenB10_tr, sBuff, 31, 14, false);

      sprintf(sBuff,"%03d", iCounter2--);
      OLEDText(&u8g2_font_ncenB10_tr, sBuff, 93, 21, false);

      sprintf(sBuff,"%03d", iCounter3--);
      OLEDText(&u8g2_font_ncenB10_tr, sBuff, 31, 31, false);
    }
    return 0;
}

int itimerTS1Check() { //Termal sensor
    int itimerTS1 = timerTS1.read()/1000;
    if(itimerTS1 > TS1_Time ) { 
     timerTS1.stop();
     itimerTS1Prev = 0;
     Serial.println("TS1 stopped.");

     Serial.print("sBuff= ");
     Serial.println(sBuff);
     Serial.println(" ");

      sprintf(sBuff,"%03d", iCounter1--);
      OLEDText(&u8g2_font_ncenB10_tr, sBuff, 31, 14, false);

      sprintf(sBuff,"%03d", iCounter2--);
      OLEDText(&u8g2_font_ncenB10_tr, sBuff, 93, 21, false);

      sprintf(sBuff,"%03d", iCounter3--);
      OLEDText(&u8g2_font_ncenB10_tr, sBuff, 31, 31, false);

     delay(20);
    }
    return 0;
}

void OLEDText(const uint8_t font, char* text, int oszlop, int sor, bool ClrBuffer) { 

  if(ClrBuffer) display.clearBuffer();
  
  display.setFont(font);
  display.setFontPosBottom();
  display.drawStr(oszlop,sor,text);
  display.sendBuffer();
}

olikraus, 2023-11-06:
The font itself is an lvalue so it doesn't make sense to calculate the address. Instead the lvalue itself should be treated as an address, so the first argument to OLEDText should be a pointer.

MPeter07, 2023-11-06:
Hi,
Many thanks, it works. :)

I modified the code:
Calling the function:
      OLEDText(u8g2_font_ncenB08_tr, sBuff, 31, 14, false);

Declaration:
void OLEDText(const uint8_t* font, char* text, int oszlop, int sor, bool ClrBuffer) { 

Best regards,
Peter

closed 2023-11-06 by MPeter07