olikraus/u8g2 #2296: Dual color OLED positioning of cursor

sizzlik, 2023-11-14:
Hi,
i recently got two ESP8266MOD boards with onboard 0.96" OLED. Im quite new to OLED displays and ran into a problem thats probably easy to fix. The OLED is one of those dual color ones with the top row in yellow and the rest blue. There was a sketch preflashed that showed the GPIO pins for the display to prove that the display is working. Now when i try to write a simple hello world it seems that im restricted to only the top row. Bottom of the text is cut off and when i setCursor a bit lower the text just starts to disapear more as if its 2 seperate displays in one. How do i "adress" the "bottom part" of the display? Here is the test code i use:

#include <Arduino.h>
#include <Wire.h>
#include<U8g2lib.h>

#define SCL 12
#define SDA 14
#define RESET U8X8_PIN_NONE
U8G2_SSD1306_128X64_NONAME_1_SW_I2C oled(U8G2_R0,SCL,SDA,RESET);

void setup(void) {
  oled.begin();
}

void loop(void) {
  oled.clearBuffer();
  oled.setFont(u8g2_font_ncenB08_tr);
  oled.drawStr(0,10,"Please Help");
  oled.sendBuffer();
  delay(1000);
}

And the result is:

[image]

How do i access the lower/blue part of the display? Thanks for any help.

olikraus, 2023-11-23:
See the "Warning" statement in the reference manual: https://github.com/olikraus/u8g2/wiki/u8g2reference#sendbuffer

This means, you must use
`U8G2_SSD1306_128X64_NONAME_F_SW_I2C oled(U8G2_R0,SCL,SDA,RESET);`

sizzlik, 2023-11-23:
> See the "Warning" statement in the reference manual: https://github.com/olikraus/u8g2/wiki/u8g2reference#sendbuffer
> 
> This means, you must use `U8G2_SSD1306_128X64_NONAME_F_SW_I2C oled(U8G2_R0,SCL,SDA,RESET);`

Thanks so much..that fixed it.

closed 2023-11-23 by sizzlik