olikraus/u8g2 #2298: SSD1363 module display error

JamesHahaho, 2023-11-18:
Hello Olikraus,

I tried an SSD1363 module (i2cinterfaced, 256x128). The panel always shows the random pattern.
[image]

Based on SSD1362 codes, what I have changed in the codes are as follow:
1. The base init sequence:
  U8X8_CA(0xfd, 0x12),            	/* unlock */
  //U8X8_C(0xAE),		                /* display off */

  U8X8_CAA(0xA0, 0x20, 0x00),
  U8X8_CA(0xA2, 0x80),    
  U8X8_CA(0xC1, 0xFF), 
  U8X8_CA(0xCA, 0x7F), 
  U8X8_CA(0xAD, 0x90), 
  U8X8_CA(0xB3, 0x61),

2.  Normal display mode command: from 0xA4 to 0xA6 （according to the datasheet）
[SSD1363.pdf](https://github.com/olikraus/u8g2/files/13399424/SSD1363.pdf)

3.  add the write RAM command in the function u8x8_d_ssd1362_common() before write data.
    u8x8_cad_SendCmd(u8x8, 0x05C);  

It seems that the control command could work properly, but the data could not write into the display RAM.

Could you kindly help on the issue?
Thanks!

James

olikraus, 2023-11-23:
I think the issue is the missing 0x05c command, which you pointed out in 3.

https://github.com/olikraus/u8g2/blob/fc312699ac4ae105ab94ad9ba9402c1cd84ffe6f/csrc/u8x8_d_ssd1362.c#L190-L193

Depending on which constructor your have used, we probably need to send the byte 0x05c as a command byte prior to line 193.
I mean we need to call `u8x8_cad_SendCmd(u8x8, 0x05c );` between lines 192 and 193.

olikraus, 2023-11-23:
Moreover the SSD1363 seems to use a different configuration for the DC flag. It must be high during argument sending, but must be low for SSD1362:
https://github.com/olikraus/u8g2/blob/fc312699ac4ae105ab94ad9ba9402c1cd84ffe6f/tools/codebuild/codebuild.c#L2056

The "cad_001" is the specification in u8g2 for this, it means for the SSD1362: 
[C]ommand: DC=0
[A]rg: DC = 0
[D]ata: DC = 1

However for the SSD1363 it sould be (according to your datasheet):
[C]ommand: DC=0
[A]rg: DC = 1
[D]ata: DC = 1

All in all the SSD1363 seems to be more similar to the SSD1322, which uses the same CAD layout and also reqires the 0x5c command for data write:

https://github.com/olikraus/u8g2/blob/fc312699ac4ae105ab94ad9ba9402c1cd84ffe6f/tools/codebuild/codebuild.c#L2031

I suggest to use an SSD1322 constructor for your display. It is worth a try.

JamesHahaho, 2023-11-24:
Dear Olikraus,
After somedays debug, I have solved the issue.

[image]

Here are the notes on the main changes:
1. to meet the requirement of D/C, I copied your `U8X_CA` series macro as `U8X8_CD`, and use them in init sequence
2. still use the `u8x8_cad_ssd13xx_fast_i2c()` as the CAD, but made changes to function `u8x8_d_ssd1363_256x128_common()` of `case U8X8_MSG_DISPLAY_DRAW_TILE` 
3. adjust the 8to32 to match the data sequence to display RAM

the .c/.h file are attached here:
[ssd1363.zip](https://github.com/olikraus/u8g2/files/13454019/ssd1363.zip)

I‘m not very familiar to the structure of U8G2, so the codes are not in u8g2 style... just for your reference.

Thanks for your hints. I'll try to rewrite the codes as your suggestions.
James

olikraus, 2023-11-24:
Thanks for the code. I will look into that once I have some time for it. I still wonder whether the ssd1322 will work...

JamesHahaho, 2023-11-26:
I tried `u8x8_cad_011` .
Due to the origin `u8x8_cad_011` dose not contain the code linked to i2c, I constructed a new function ` u8x8_cad_H011()` :
```
uint8_t u8x8_cad_H011(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{ 
  switch(msg)
  {
    case U8X8_MSG_CAD_SEND_CMD:
    printf("c");
      u8x8_byte_SetDC(u8x8, 0);
      u8x8_byte_SendByte(u8x8, arg_int);
      break;
    case U8X8_MSG_CAD_SEND_ARG:
    printf("a");
      u8x8_byte_SetDC(u8x8, 1);
      u8x8_byte_SendByte(u8x8, arg_int);
      break;
    case U8X8_MSG_CAD_SEND_DATA:
    printf("d");
      u8x8_byte_SetDC(u8x8, 1);
      u8x8_byte_SendBytes(u8x8, arg_int, arg_ptr);
      break;
      /* fall through */
    case U8X8_MSG_CAD_INIT:        /*added codes here*/
          if ( u8x8->i2c_address == 255 )
	      u8x8->i2c_address = 0x078;
        break;
    case U8X8_MSG_CAD_START_TRANSFER:
    case U8X8_MSG_CAD_END_TRANSFER:
      return u8x8->byte_cb(u8x8, msg, arg_int, arg_ptr);
    default:
      return 0;
  }
  return 1;
}
```
The data transmit on i2c bus looks like :
[image]
It does not work.

Compare to the worked codes, the transmitted data look like:
[image]

It seems the command+arg series could not be transmitted in one i2c package.

The logic_analyzer data is attached for your reference. You can use [DSView](https://dreamsourcelab.cn/download/) to view them.
[logic_analizer_data.zip](https://github.com/olikraus/u8g2/files/13467042/logic_analizer_data.zip)

olikraus, 2023-11-26:
The cad_001 / cad011 are indeed only for SPI interface. Maybe i was confused here, but the u8x8_cad_ssd13xx_fast_i2c() should be correct for I2C.

musikkfeil, 2024-03-26:
I still don’t understand how to connect SSD1363 I2S. Is this possible and how?

neuralmax, 2024-04-02:
Can you please post the logic_analyzer data for write to RAM sequence? I has been weeks and I still cannot get it to work.

JamesHahaho, 2024-04-03:
[la_data.zip](https://github.com/olikraus/u8g2/files/14845054/la_data.zip)
Here I re-attached the logic analyzer data for accessing SDD1363 with I2C. 
You may use DSLogic software (https://dreamsourcelab.cn/download/) to open it.

neuralmax, 2024-04-13:
Thank you so much. I managed to run it at last. For now just using the Wire library and sending the codes I copied from your logic analysis files. Although it works its seems very slow, despite that I run I2C at 1HMz clock speed and managed to send whole line in one communication burst I get only about 2 fps and frame tearing. Is it slow like this for you also? Link to video of result I got: https://youtu.be/XztWQ6t6XSM

thedonalddon, 2024-12-10:
Thank you so much. I managed to run it at last. For now just using the Wire library and sending the codes I copied from your logic analysis files. Although it works, it seems very slow. Despite running I2C at 1MHz clock speed and managing to send the entire line in one communication burst, I only get about 2 fps and frame tearing. Is it slow like this for you also? Link to video of result I got: https://youtu.be/XztWQ6t6XSM

I have the same problem as you. It's working, but has a very slow refresh rate.  Each time it refreshes, it is divided into several lines, and each line is about 10 pixels high.

thedonalddon, 2024-12-10:
> Thank you so much. I managed to run it at last. For now just using the Wire library and sending the codes I copied from your logic analysis files. Although it works its seems very slow, despite that [...]

I have the same problem as you. It's working, but has a very slow refresh rate. Each time it refreshes, it is divided into several lines, and each line is about 10 pixels high.

olikraus, 2024-12-11:
I assume I can't help much from a u8g2 point of view...

gabormull, 2025-01-20:
Dear Olikraus,
I would like to know if you managed to find a solution for controlling the SSD1363 (I2C interfaced)?
I tried the above solutions, but I didn't get the desired result.

olikraus, 2025-01-20:
No progress from my end, but I probably should add this to the next release...

olikraus, 2025-01-20:
There actually is code for the SSD1363: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1363.c
Indeed there should be a constructor for I2C:  
https://github.com/olikraus/u8g2/blob/2985691bbf473f35a0c341cc52b2837320d2bb7d/tools/codebuild/codebuild.c#L2289-L2295

There should be a constructor `U8G2_SSD1363_256X128_1_HW_I2C`.
Can you do a test with the same and send the output here?

gabormull, 2025-01-20:
Yes, of course. I'll report back tomorrow with the results.

gabormull, 2025-01-21:
[image]
I completed the requested test, but unfortunately there are no good results.

olikraus, 2025-01-22:
I think the I2C procedure is wrong, new version probably is this (CAD=011 instead of 001):
``` C++
uint8_t u8x8_cad_011_ssd13xx_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t *p;
  switch(msg)
  {
    case U8X8_MSG_CAD_SEND_CMD:
      u8x8_byte_StartTransfer(u8x8);
      u8x8_byte_SendByte(u8x8, 0x000);
      u8x8_byte_SendByte(u8x8, arg_int);
      u8x8_byte_EndTransfer(u8x8);      
      break;
    case U8X8_MSG_CAD_SEND_ARG:
      u8x8_byte_StartTransfer(u8x8);
      u8x8_byte_SendByte(u8x8, 0x040);
      u8x8_byte_SendByte(u8x8, arg_int);
      u8x8_byte_EndTransfer(u8x8);      
      break;
    case U8X8_MSG_CAD_SEND_DATA:
      p = (uint8_t *)arg_ptr;
       while( arg_int > 24 )
      {
	u8x8_i2c_data_transfer(u8x8, 24, p);
	arg_int-=24;
	p+=24;
      }
      u8x8_i2c_data_transfer(u8x8, arg_int, p);
      break;
    case U8X8_MSG_CAD_INIT:
      /* apply default i2c adr if required so that the start transfer msg can use this */
      if ( u8x8->i2c_address == 255 )
	u8x8->i2c_address = 0x078;
      return u8x8->byte_cb(u8x8, msg, arg_int, arg_ptr);
    case U8X8_MSG_CAD_START_TRANSFER:
    case U8X8_MSG_CAD_END_TRANSFER:
      /* cad transfer commands are ignored */
      break;
    default:
      return 0;
  }
  return 1;
}
```

olikraus, 2025-01-22:
@gabormull : I fixed 
U8G2_SSD1363_256X128_1_HW_I2C
U8G2_SSD1363_256X128_1_SW_I2C
U8G2_SSD1363_256X128_F_HW_I2C
U8G2_SSD1363_256X128_F_SW_I2C

Can you test beta release 2.36.4 with one of the above constructors? 

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

gabormull, 2025-01-23:
[image]
It works perfectly. Thank you very much.

olikraus, 2025-01-23:
nice

rhoggarth, 2025-03-17:
Hi - I'm using the beta 2.36.5 on an ESP32. Good news is the HW and SW I2C do drive the display correctly, but the performance is super slow.  SW about twice as slow as HW as would be expected, but even HW takes 1/2 second to refresh the screen. I'm moving from an ssd1309 to the ssd1363 because of increased resolution. ssd1309 performance is instantaneous compared to 1363.

olikraus, 2025-03-26:
Oh.... Maybe there is a difference between AVR and ESP systems...

olikraus, 2025-03-26:
BTW: Do you mean setBusClock() ?

rhoggarth, 2025-03-26:
Espressif documentation doesn't say whether Wire.setClock() should be before or after the Wire.begin().  I just tried and putting it after works, but putting it before does not (defaults to 100kHz).
`#define W_BEGIN(sda, scl) Wire.setClock(1000000);Wire.begin(sda, scl) // does not work`
but
`#define W_BEGIN(sda, scl) Wire.begin(sda, scl);Wire.setClock(1000000) // works`