olikraus/u8g2 #2299: Uppercase u8g2_font_helvR08_tr letters are offset by 1

harbaum, 2023-11-18:
I am writing a file selector using u8g2/mui and I noticed, that many uppercase letters of the font u8g2_font_helvR08_tr are offset by one pixel to the right as if they had an empty first pixel column as seen with the uppercase G and H which appear one pixel shifted to the right.

[image]

This does not happen with the bold variant of that font. 

Is this a bug in the font itself or in the conversion?

olikraus, 2023-11-23:
lower case "H":
https://github.com/olikraus/u8g2/blob/fc312699ac4ae105ab94ad9ba9402c1cd84ffe6f/tools/font/bdf/helvR08.bdf#L618-L622

lower case "h":
https://github.com/olikraus/u8g2/blob/fc312699ac4ae105ab94ad9ba9402c1cd84ffe6f/tools/font/bdf/helvR08.bdf#L1081-L1085

The third value of the BBX command is the x offset from the left.
So: The offset is part of the orginal specification.

harbaum, 2023-11-23:
Oh wow. I wonder what's the purpose  of that. It doesn't only look odd, it also wastes a few precious pixels.

I think I'll try to remove all leading offsets and give the resulting font a try and report here if that has some unexpected side effects.

olikraus, 2023-11-23:
I don't remember details, but maybe this function helps:

https://github.com/olikraus/u8g2/blob/fc312699ac4ae105ab94ad9ba9402c1cd84ffe6f/csrc/u8g2.h#L1783

It should return 1 for u8g2_GetStrX(u8g2.GetU8g2(), "Hello");	
and should return 0 for u8g2_GetStrX(u8g2.GetU8g2(), "hello");	

sooo... if you want to print string s at position x, then instead print s at position

x - u8g2_GetStrX(u8g2.GetU8g2(), s);	

well, as mentiond, this is just an idea and not tested.

hmmm... i think there should be an old Atari ST in my parents house :-D

olikraus, 2023-11-23:
Just refering to a similar topic: https://github.com/olikraus/u8g2/issues/1561

harbaum, 2023-11-23:
> hmmm... i think there should be an old Atari ST in my parents house :-D
You won't need it anymore. I am building a cheap, small and with modern peripherals compatible replacement.

Anyway, I just tried to rebuild the fonts and the build process fails with 

```
Can not open bdf file 'tmp.bdf'
font ../ttf/Minimal3x3.ttf missing
font ../ttf/Minimal3x3.ttf missing
update u8g2.h
update u8x8.h
Segmentation fault (core dumped)
```

And indeed, there is no `../ttf/Minimal3x3.ttf` ... should there be one?

harbaum, 2023-11-23:
Out of curiosity, I've modified the BDF parser like this:

```
  /* copy the values from the parser to the glyph record */
  bg->encoding = bf->encoding;
  bg->dwidth_x = bf->dwidth_x;
  bg->dwidth_y = bf->dwidth_y;
  bg->bbx.w = bf->bbx_w;
  bg->bbx.h = bf->bbx_h;
  bg->bbx.x = bf->bbx_x;
  bg->bbx.y = bf->bbx_y;

  if(bg->bbx.x > 0) {
    bg->dwidth_x -= bg->bbx.x;
    bg->bbx.x = 0;
  }
```

which got me from this:  

[image]

to this:

[image]

IMHO, this looks much nicer. I could probably do the same when rendering letter-by-letter and doing the x adjustment similar to what you suggested for each letter.

But maybe I should just use a custom and modified version of helvR08 ... This would also allow me to reduce the digit spacing. Currently, they are all the same width which adds extra space behind e.g. the digit '1'.

harbaum, 2023-11-23:
Please don't hesitate to drop me a line if you'd like support reviving your Atari ST memories with MiSTeryNano ...

closed 2023-11-23 by harbaum

olikraus, 2023-11-27:
> And indeed, there is no ../ttf/Minimal3x3.ttf ... should there be one?

strange, I will check this. Maybe I missed to add this to the repo :-o
ok, it wasn't there... not sure why, but I added the ttf to the repo.

> https://www.youtube.com/shorts/zsHYcolqtpc

very nice