olikraus/u8g2 #2303: input from matrix keypad to u8g2.userInterfaceSelectionList()

sudheerbussa, 2023-11-22:
I am using a matrix keypad of 4X3. if i want to use keys to navigate through menu, how to pass arguments from to this function from keypad instead of direct GPIO connections. I am using keypad.h to read key from matrix keypad.  Your help is appreciated. Thank you

olikraus, 2023-11-22:
userInterface procedures are more like a small demo thing instead of a full featured flexible user interface. I suggest to use https://github.com/neu-rah/ArduinoMenu or https://github.com/olikraus/u8g2/wiki/muimanual

sudheerbussa, 2023-11-23:
Thank you for the response. Is there any way to use 4X4 matrix keypad output values as arguments of the u8g2.begin() function instead of using individual pins. please note i am using keypad.h to read key from matrix keypad.
`u8g2.begin(/* menu_select_pin= */ 5, /* menu_next_pin= */ 4, /* menu_prev_pin= */ 2, /* menu_up_pin= */ U8X8_PIN_NONE, /* menu_down_pin= */ U8X8_PIN_NONE, /* menu_home_pin= */ 3); `

olikraus, 2023-11-23:
No, this is out of the scope for u8g2. select a suitable lib by yourself and combine it with MUI.

sudheerbussa, 2023-11-23:
It seems i am getting there as for as understanding MUI is concerned as it can handle sosftware events along with GPIO . I have a question. As most of the examples in MUI initialized display in page mode, does it support both page and Fullbuffer mode?

olikraus, 2023-11-23:
> does it support both page and Fullbuffer mode?
Yes of course.