olikraus/u8g2 #2304: JLX12864G-1016

plutphil, 2023-11-26:
I'm trying to get the lcd display JLX12864G-1016 to work. It kind of works with the drivers ST7565, ST7588 and UC1701, but it shows the hello world text, i'm trying to print only in a blink of an eye during reset, init.
I wrote a bit bang software driver based on the chinese datasheet pdf  that works. [http://www.jlxlcd.cn/UpFile/201461011200.pdf](http://www.jlxlcd.cn/UpFile/201461011200.pdf). I translated the pdf for the comments in this code. How do i get this library to work with this display?

```
digitalWrite(PIN_RES,0);
delay(100);
digitalWrite(PIN_RES,1);
delay(20);
sendCommand(0xe2);delay(5);//soft reset
sendCommand(0x2c);delay(5);//boot step 1 Power control Select internal voltage supply operating mode
sendCommand(0x2e);delay(5);//boot step 2
sendCommand(0x2f);delay(5);//boot step 3

sendCommand(0x24);/*Coarse contrast adjustment, settable range 0x20~0x27*/
sendCommand(0x81);/*Fine-tuning the contrast*/
sendCommand(0x1a);/*0x1a, fine-tune the contrast value, the settable range is 0x00~0x3f*/
sendCommand(0xa2);/*1/9 bias ratio (bias)*/
sendCommand(0xc8);/*Row scanning order: from top to bottom*/
sendCommand(0xa0);/*Column scanning order: from left to right*/

sendCommand(0x40);/*Starting line: starts from the first line*/
sendCommand(0xaf);/*Open display/display on (0xae off)*/
digitalWrite(PIN_CS,1); 

void lcdAddr(int page = 0, int column = 0){
  digitalWrite(PIN_CS,0);
  delayMicroseconds(50);
  //delay(1);
  column--;
  page--;
  sendCommand(0xb0+page);
  sendCommand(((column>>4)&0x0f)+0x10);
  sendCommand(column&0x0f);
}

for (size_t i = 0; i < 9; i++)
{
  lcdAddr(1+i, 1);
  for (size_t j = 0; j < 132; j++)
  {
    sendData(rand());
  }
}
```

plutphil, 2023-11-26:
I got it working now with this driver and hardware spi. I'm using an esp32 c3. The spi frequency works fine all the way up to 80MHz. I noticed that i get an image, but the contrast was too low. So i resend the commands adjusting the contrast manually after begin. 
U8G2_UC1701_EA_DOGS102_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 0, /* dc=*/ 2, /* reset=*/ 1);

SPI.begin(4, 5, 3, 0);
SPI.setFrequency(80 * 1000 * 1000);
u8g2.begin();
u8g2.sendF("c",0x24);
u8g2.sendF("c",0x81);
u8g2.sendF("c",0x1a);
u8g2.sendF("c",0xa2d);

[image]

Now the text isn't properly aligned.

plutphil, 2023-11-26:
Can i reduce the contrast to make the display refresh faster?

olikraus, 2023-11-26:
Did you try the UC1701 128x64 constructor `U8G2_UC1701_MINI12864_F_4W_HW_SPI`? https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#uc1701-mini12864

The contrast can be adjusted with u8g2.setContrast() https://github.com/olikraus/u8g2/wiki/u8g2reference#setcontrast

The SPI frequency should be adjusted with u8g2.setBusClock https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock

plutphil, 2023-11-26:
Thanks it works now. How can i invert the colors?

olikraus, 2023-11-26:
https://github.com/olikraus/u8g2/wiki/u8g2reference#setdrawcolor

closed 2023-11-26 by plutphil