olikraus/u8g2 #2306: characters glitch

matiasbarone, 2023-11-28:
Im ussing the library with ESP32-S2 and Arduino.
Im getting glitchs in some characters, im checking them before calling to the u8g2 functions and the text is OK

This is the glitch im getting (4th character second line):
https://github.com/olikraus/u8g2/assets/5585590/c14ea60a-c87d-44e9-9eb3-3516e24d6f85

The most of the times it is on the same characters, but not always. Im calling `begin` every 30 seconds but still have this glitch.

Im initializig with: 
`U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 35, 37, 36);`

Thanks for your help

olikraus, 2023-11-29:
Looks like a hardware issue. The ST 7920 is very sensitive here. Did you use proper 5V data lines for this display?

matiasbarone, 2023-11-29:
> Looks like a hardware issue. The ST 7920 is very sensitive here. Did you use proper 5V data lines for this display?

No, I'm going to try that